ZJU-FAST-Lab/ego-planner #1: 在replan的时候，优化结果始终是lbfgs::LBFGSERR_CANCELED，什么情况会导致终止最小化进程？

IngaVS, 2020-08-18:
[image]
在replan的时候，优化结果始终是lbfgs::LBFGSERR_CANCELED，什么情况会导致终止最小化进程？
此外，replan设置start_pt_的时候不需要判断当前时间是否在轨迹时间内吗？

bigsuperZZZX, 2020-08-18:
当轨迹发现新障碍物的时候会给优化器发出命令，提前退出当前优化并令求解器返回lbfgs::LBFGSERR_CANCELED，然后加入新的障碍物信息，重新启动优化。
至于为啥一直返回lbfgs::LBFGSERR_CANCELED，要不截个rviz的图看看此时飞机和目标点在哪？图里描述的状态很奇怪诶。
问题二：要判断的，如果当前时间超出轨迹时间，状态机就切换到等待目标点了。

bigsuperZZZX, 2020-08-19:
如果没啥问题了，请尽早关闭这个issue，以及以后还是用英语交流吧

closed 2020-08-19 by IngaVS

w407022008, 2022-05-29:
Hello, I have the same issue shown in the following pictures. It worked well if I modified the target point so that there were almost no obstacles in a straight line. But if such an obstacle existed between it would have such issue. So what's the key point?
I used the default parameters (the obstacle inflation parameter was set to 0.25, and 3dLidar plugin was used here to provide point clouds) and demanded the drone to go around a column with a diameter of 0.8m. The vehicle and the target were almost both 1.5m from the centre of the obstacle. 
[image]
[image]

w407022008, 2022-06-02:
Fixed. Anyway, we just need to tune those opt parameters.