ZJU-FAST-Lab/ego-planner #2: Issue when setting up package

kadhirumasankar, 2020-09-02:
Hi. I cloned the repo and ran `catkin_make -DCMAKE_BUILD_TYPE=Release`. When I do that, I get the following error:
```bash
...
[ 84%] Built target waypoint_generator
[ 84%] Building CXX object uav_simulator/Utils/odom_visualization/CMakeFiles/odom_visualization.dir/src/odom_visualization.cpp.o
In file included from /opt/ros/melodic/include/ros/node_handle.h:32:0,
                 from /opt/ros/melodic/include/ros/ros.h:45,
                 from /home/kadhir/catkin_ws/src/ego-planner/src/uav_simulator/Utils/odom_visualization/src/odom_visualization.cpp:3:
/opt/ros/melodic/include/ros/publisher.h:53:52: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
     Publisher& operator=(const Publisher& other) = default;
                                                    ^
In file included from /opt/ros/melodic/include/ros/node_handle.h:33:0,
                 from /opt/ros/melodic/include/ros/ros.h:45,
                 from /home/kadhir/catkin_ws/src/ego-planner/src/uav_simulator/Utils/odom_visualization/src/odom_visualization.cpp:3:
/opt/ros/melodic/include/ros/subscriber.h:52:52: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   Subscriber& operator=(const Subscriber& other) = default;
                                                    ^
In file included from /opt/ros/melodic/include/ros/node_handle.h:34:0,
                 from /opt/ros/melodic/include/ros/ros.h:45,
                 from /home/kadhir/catkin_ws/src/ego-planner/src/uav_simulator/Utils/odom_visualization/src/odom_visualization.cpp:3:
/opt/ros/melodic/include/ros/service_server.h:51:58: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   ServiceServer& operator=(const ServiceServer& other) = default;
                                                          ^
In file included from /opt/ros/melodic/include/ros/node_handle.h:35:0,
                 from /opt/ros/melodic/include/ros/ros.h:45,
                 from /home/kadhir/catkin_ws/src/ego-planner/src/uav_simulator/Utils/odom_visualization/src/odom_visualization.cpp:3:
/opt/ros/melodic/include/ros/service_client.h:49:58: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   ServiceClient& operator=(const ServiceClient& other) = default;
                                                          ^
In file included from /opt/ros/melodic/include/ros/node_handle.h:36:0,
                 from /opt/ros/melodic/include/ros/ros.h:45,
                 from /home/kadhir/catkin_ws/src/ego-planner/src/uav_simulator/Utils/odom_visualization/src/odom_visualization.cpp:3:
/opt/ros/melodic/include/ros/timer.h:52:42: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   Timer& operator=(const Timer& other) = default;
                                          ^
In file included from /opt/ros/melodic/include/ros/node_handle.h:38:0,
                 from /opt/ros/melodic/include/ros/ros.h:45,
                 from /home/kadhir/catkin_ws/src/ego-planner/src/uav_simulator/Utils/odom_visualization/src/odom_visualization.cpp:3:
/opt/ros/melodic/include/ros/wall_timer.h:52:50: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   WallTimer& operator=(const WallTimer& other) = default;
                                                  ^
In file included from /opt/ros/melodic/include/ros/node_handle.h:39:0,
                 from /opt/ros/melodic/include/ros/ros.h:45,
                 from /home/kadhir/catkin_ws/src/ego-planner/src/uav_simulator/Utils/odom_visualization/src/odom_visualization.cpp:3:
/opt/ros/melodic/include/ros/steady_timer.h:52:54: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
   SteadyTimer& operator=(const SteadyTimer& other) = default;
                                                      ^
In file included from /opt/ros/melodic/include/tf/transform_broadcaster.h:36:0,
                 from /home/kadhir/catkin_ws/src/ego-planner/src/uav_simulator/Utils/odom_visualization/src/odom_visualization.cpp:4:
/opt/ros/melodic/include/tf/tf.h:354:8: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
   std::shared_ptr<tf2_ros::Buffer> getTF2BufferPtr() { return tf2_buffer_ptr_;};
        ^
/opt/ros/melodic/include/tf/tf.h:403:8: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
   std::shared_ptr<tf2_ros::Buffer> tf2_buffer_ptr_;
        ^
/opt/ros/melodic/include/tf/tf.h: In member function ‘ros::Duration tf::Transformer::getCacheLength()’:
/opt/ros/melodic/include/tf/tf.h:331:43: error: ‘tf2_buffer_ptr_’ was not declared in this scope
   ros::Duration getCacheLength() { return tf2_buffer_ptr_->getCacheLength();}
                                           ^
/opt/ros/melodic/include/tf/tf.h: In member function ‘void tf::Transformer::setUsingDedicatedThread(bool)’:
/opt/ros/melodic/include/tf/tf.h:349:46: error: ‘tf2_buffer_ptr_’ was not declared in this scope
   void setUsingDedicatedThread(bool value) { tf2_buffer_ptr_->setUsingDedicatedThread(value);};
                                              ^
/opt/ros/melodic/include/tf/tf.h: In member function ‘bool tf::Transformer::isUsingDedicatedThread()’:
/opt/ros/melodic/include/tf/tf.h:351:42: error: ‘tf2_buffer_ptr_’ was not declared in this scope
   bool isUsingDedicatedThread() { return tf2_buffer_ptr_->isUsingDedicatedThread();};
                                          ^
uav_simulator/Utils/odom_visualization/CMakeFiles/odom_visualization.dir/build.make:62: recipe for target 'uav_simulator/Utils/odom_visualization/CMakeFiles/odom_visualization.dir/src/odom_visualization.cpp.o' failed
make[2]: *** [uav_simulator/Utils/odom_visualization/CMakeFiles/odom_visualization.dir/src/odom_visualization.cpp.o] Error 1
CMakeFiles/Makefile2:4424: recipe for target 'uav_simulator/Utils/odom_visualization/CMakeFiles/odom_visualization.dir/all' failed
make[1]: *** [uav_simulator/Utils/odom_visualization/CMakeFiles/odom_visualization.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 85%] Building CXX object uav_simulator/Utils/rviz_plugins/CMakeFiles/rviz_plugins.dir/src/goal_tool.cpp.o
...
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```

How do I fix this error? I removed the code that I didn't think was relevant. Please let me know if this is sufficient information.

bigsuperZZZX, 2020-09-03:
It seems that an old C++ version is used to compile the code by default. The minimum version required is C++11. Maybe specifying the version explicitly can solve the problem.
You can add the following code 
```
ADD_COMPILE_OPTIONS(-std=c++11 ) 
ADD_COMPILE_OPTIONS(-std=c++14 )
```
to the [CMakeLists.txt](https://github.com/ZJU-FAST-Lab/ego-planner/blob/master/src/uav_simulator/Utils/odom_visualization/CMakeLists.txt) of the package _odom_visualization_ at line 41, like 
[image].
Hoping this can help.

kadhirumasankar, 2020-09-04:
Thanks a lot! That fixed the error I was getting during `catkin_make`, but when running `roslaunch ego_planner rviz.launch` and `roslaunch ego_planner run_in_sim.launch`, RViz opens up as it's supposed to, but I don't see the drone, or a colored visualization like in your GIF. I can still use 2D Nav Goal to publish points though. This is what I see:
[image]

I get this warning when roslaunching run_in_sim:
```
[odom_visualization-7] process has died [pid 7157, exit code -11, cmd /home/kadhir/catkin_ws/src/ego-planner/devel/lib/odom_visualization/odom_visualization ~odom:=/visual_slam/odom __name:=odom_visualization __log:=/home/kadhir/.ros/log/f6380648-ee48-11ea-a97a-9453307180b5/odom_visualization-7.log].
log file: /home/kadhir/.ros/log/f6380648-ee48-11ea-a97a-9453307180b5/odom_visualization-7*.log
```

bigsuperZZZX, 2020-09-04:
Are you using a virtual machine? I encountered a similar error when testing in a virtual machine.
What's more, it seems that package _odom_visualization_, which is just for visualization(The red drone model in GIFs),  has died. So if you are unable to fig it out, you can ignore this _odom_visualization_ package and display the position by yourself from an odometry topic.

kadhirumasankar, 2020-09-04:
I'm not using a virtual machine. Yes, I think you are right that the `odom_visualization` package failed, that seems to be reflected in the warning I got too. Would you happen to know how to fix that? I don't have any experience with C++, so I'm not sure what to do.

bigsuperZZZX, 2020-09-05:
The error you reported reveals nothing to the reason for the failure. Package _odom_visualization_ has never failed in my computer and different platforms of onboard computers on the drones. 
Maybe you can follow the commands 
```
sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt-get update
sudo apt-get install gcc-7 g++-7
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-5 60 --slave /usr/bin/g++ g++ /usr/bin/g++-5
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-7 50 --slave /usr/bin/g++ g++ /usr/bin/g++-7
```
to reconfigure the compiler, then delete _build/_ and _devel/_ folder and recompile the code. 
Hoping this helps. If you still get the same error, maybe I'm not able to provide more advice.

kadhirumasankar, 2020-09-14:
I ran all those steps, but I still get the same error as before saying `odom_visualization` has died. Not sure what the problem is. Just to confirm, do you think you'll be to provide any more help?

bigsuperZZZX, 2020-09-15:
I'm sorry that I can't fix the `odom_visualization` error you encounted.

kadhirumasankar, 2020-09-17:
No worries. Thank you for your help

closed 2020-09-17 by kadhirumasankar