ZJU-FAST-Lab/ego-planner #4: Importing Real environment

Amit10311, 2020-09-08:
Thank you for this great package for simulating MAV. I am running Ubuntu 16.04. The simulator is working perfectly. I want to set up my own environment. You have created the environment with the use of a `mock map` package. I want to import the .dae environment structure.  Can you please guide me on how to set up the environment in your package.

I look forward to hearing from you. Thank you for your valuable time.

bigsuperZZZX, 2020-09-09:
It's easy to do this. Use command `rqt_graph` to see the nodes and active topics
[image]
The node _/mockamap_node_ at the left bottom of the figure is the node generating the map.
You can just to replace _/mockamap_node_ with your own map generating node publishing you map topic named _/map_generator/global_cloud_ of type _sensor_msgs/PointCloud2_ at a rate of every one or two seconds.

Amit10311, 2020-09-15:
Thank you for your reply. I would like to mention that when  I am publishing the `pointclou2data` it's not visible in the `rviz.launch` but it's visible in the normal rviz.

bigsuperZZZX, 2020-09-15:
Make sure that you have renamed your pointcloud topic to "/map_generator/global_cloud".
What's more, don't forget to disable the original _/mockamap_node_ .

Amit10311, 2020-09-16:
In the map generating you are having these topics :
```
Node [/random_map_sensing]
Publications: 
 * /map_generator/global_cloud [sensor_msgs/PointCloud2]
 * /map_generator/local_cloud [sensor_msgs/PointCloud2]
 * /pcl_render_node/local_map [sensor_msgs/PointCloud2]

Subscriptions: 
 * /random_map_sensing/odometry [unknown type]
```
The point cloud data I am providing is prebuilt data. So, basically I am providing  `/map_generator/global_cloud [sensor_msgs/PointCloud2]`. While providing this data it shows:
[ WARN] [1600242929.182409695]: Global Pointcloud received.

Can you please guide me on how to set up the pre-built data in your package.

bigsuperZZZX, 2020-09-18:
`[ WARN] [1600242929.182409695]: Global Pointcloud received.` means the global map has successfully received by the planner and everything work well. We use ROS_WARN(.) is just to let this message displayed in yellow, so that we can distinguish it from numerous message flow easier from termianl. 
You can publish the map only once, and be sure the planner has received the map when this "yellow message" is seen.
Our package has no built-in function for publishing pre-built data. However, any method that converts your .dae file to a ROS message is OK.

Amit10311, 2020-09-18:
The cloud map is still not visible. My map is big, is there a limit of map dimension? 

I also like to know in which node the `click goals` are generated. I would like to choose the `click goal points` at high altitude. Could you please provide the information regarding it?

bigsuperZZZX, 2020-09-28:
If there is no more questions, please close this issue, thanks.

closed 2020-09-29 by Amit10311