ZJU-FAST-Lab/ego-planner #5: some question about time initialization in function reboundReplan (planner_manager.cpp)

tiemuhua, 2020-09-20:
[image]
it seems like you treat the motion with start velocity zero and constant acceleration MAX_ACC. 
i don't know why this estimation of cost time is reasonable, because the start velocity do affect the time needed largely.
Moreover, if we do treat the motion as a uniform acceleration motion, perhaps the denominator should be 2*MAX_ACC, as X=V^2/(2*a) for uniform acceleration motion.

bigsuperZZZX, 2020-09-20:
I couldn't find this code in my project, could tell me where it lies in?
But I found a similar part in my recent code in _planner_manager.cpp_.
the '2' at denominator was reduced because we assume the trajectory starts and ends both under zero velocity, and there are two parts of acceleration and deceleration. In fact this naive assumption only slightly enlarges the total time allocation, which will not do significant affect to the final trajectory since the trajectory optimizer will generate a satisfied trajectory later.

tiemuhua, 2020-09-20:
> I couldn't find this code in my project, could tell me where it lies in?
> But I found a similar part in my recent code in _planner_manager.cpp_.
> the '2' at denominator was reduced because we assume [...]

well,  in the original project there is a function called reboundReplan which i think is too too too long,  especially the first part (the init part),  so i decided the init part into to function: initByPreTraj and initByMinSnap. 
Thanks for your answer

bigsuperZZZX, 2020-09-20:
You are welcome.

closed 2020-09-20 by bigsuperZZZX