ZJU-FAST-Lab/ego-planner #6: [ego_planner_node-1] process has died

lovZgump, 2020-09-23:
Thank you for this great work. I am running Ubuntu 16.04. But I can't  run the planner in simulation. When I launch the **run_in_sim.launch**, the process always died. I haven't change anything. Can you please guide me on how to fix the problem.

I am looking forward to hearing from you. Thank you for your valuable time.
[image]

bigsuperZZZX, 2020-09-23:
We encountered a similar problem(not exactly the same) and it was proved to be a BUG. This occurs on a computer with limited computing resources. We fixed this bug but has not been committed. So we commit it. We added the following code in _ego_planner_fsm.cpp_, so you can manually modify the code as the following picture or re-download the repository. Hoping this will solve you problem.
[image]

lovZgump, 2020-09-23:
Thank you for your reply. The problem has been fixed. Thanks again.

closed 2020-09-23 by lovZgump