ZJU-FAST-Lab/ego-planner #7: some questions about function initControlPoints in file bspline_optimizer.cpp

tiemuhua, 2020-09-23:
[image]
MINIMUM_PERCENT is 0.0, so minimum_points should be 0, so the bool expression in line 413 should always be false, what is the purpose of these codes?

bigsuperZZZX, 2020-09-23:
In some situations in the early time, we find that one single control point is unable to produce enough repulsive force. One effective way is to put some force onto adjacent points to make the trajectory producing stronger force.
The parameter MINIMUM_PERCENT specified how much control points will be assigned force for one obstacle. But we set this to 0, which still works OK...

tiemuhua, 2020-09-23:
> In some situations in the early time, we find that one single control point is unable to produce enough repulsive force. One effective way is to put some force onto adjacent points to make the [...]

so do you means that line 413 to line 422 can be removed?

bigsuperZZZX, 2020-09-23:
It depends on the environment you want to fly in. If the environment contains many big obstacles, we recommend you to keep this code and set the MINIMUM_PERCENT to 10.0 ~ 20.0 if the planner performs unsatisfactory. Or this code can be removed.

bigsuperZZZX, 2020-09-28:
If there is no more questions, please close this issue, thanks.

closed 2020-09-28 by tiemuhua