ZJU-FAST-Lab/ego-planner #8: How do I send a new waypoint without using RViz?

kadhirumasankar, 2020-09-26:
I'm trying to set up a rospy node that I can use to set a new waypoint for the drone to go to. What topic should I publish to?

I tried `ego_planner_node/goal_point` and `move_base_simple/goal` and I got an error saying those nodes hadn't been initialized yet. When I tried to initialize them, I got an error saying namespaces could not be used in rosnode initializations.

bigsuperZZZX, 2020-09-26:
Well, it's easy. I recommend you to read the following code in [ego_replan_fsm.cpp](https://github.com/ZJU-FAST-Lab/ego-planner/blob/master/src/planner/plan_manage/src/ego_replan_fsm.cpp) first, which is related to waypoint receiving and processing.
[image]
[image]
[image]
Then I'm sure you can figure out how to send custom waypoints.

bigsuperZZZX, 2020-09-28:
If there is no more questions, please close this issue, thanks.

kadhirumasankar, 2020-09-29:
Hi. Thanks for the reply

I went through the file you mentioned, and I noticed that the trajectory planner is subscribing to the `/waypoint_generator/waypoints` topic. So, in my rospy node, I am trying to publish a Path message to that topic. 

```python
waypoint_pub = rospy.Publisher('/waypoint_generator/waypoints', Path, queue_size=1)
waypoint_pub.publish("hello") # just passing hello for now to make sure it works
```

When I run that, I get the following error:
```bash
File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 878, in publish
    raise ROSException("ROS node has not been initialized yet. Please call init_node() first")
rospy.exceptions.ROSException: ROS node has not been initialized yet. Please call init_node() first
```

This is the same error I got as last time. On projects in the past, I have published messages in a similar way, but I never got this error before. How do I solve this? Please note that I am working with Python, not C++.

kadhirumasankar, 2020-09-29:
Hi. Thanks for the reply, and thanks for referring me to that code. 

For the reference of anyone who looks at this issue, the trajectory planner is subscribing to the `/waypoint_generator/waypoints` topic. I have not tried this yet, but I expect that new messages must be published to that topic.

closed 2020-09-29 by kadhirumasankar