ZJU-FAST-Lab/ego-planner #9: planning without considering the current odometry

zhengpw33, 2020-09-27:
Why is line 294 in file ego_replan_fsm.cpp using time duration to get the new control positon, without considering the current odometry? 
By the way，I’ m wondering that the fig 4(a) in paper shows the trajectory passes all of the control points（Q） of B-spline，is that clerical error？

bigsuperZZZX, 2020-09-27:
Thank you for detailed reading.

For the question 1: The position we get at line 294 in file ego_replan_fsm.cpp is just for determining whether to start a new replan, not for control or other purpose. So it doesn't need to be accurate and whatever the source is, it’s OK.

For the question 2: My professor once discussed this issue with me before submission as well. And we choose to let the trajectory pass through the series of points for two reasons: 1. the trajectory will get very close to each control points if the placement of the control points is "smooth" (the integral of the acceleration/jerk of the trajectory represented by the control point is small). So in order to simplify the illustration, we make the trajectory pass through these points. 2. Although we use control points as the decision variables, our front-end obstacle information extraction method is not restricted to B-spline parameterization. Any kind of waypoint representation method is compatible. So passing through these points is more general.

Bruno-33, 2020-09-27:
> Thank you for detailed reading.
>
> For the question 1: The position we get at line 294 in file ego_replan_fsm.cpp is just for determining whether to start a new replan, not for control or other [...]

Thank you for your rely. But I'm still confused! I‘m try to run the egp_planner to get the trajectory in a obstacle-free map,but the trajectory is updating and moving toward to the goal even though the current odometry is static.How can I get the trajectory considering the current odometry?

bigsuperZZZX, 2020-09-27:
That's true, we decoupled trajectory generation and trajectory tracking control. This is reasonable in most cases(no significant 
 and constant external disturbance) since the trajectories are generated under kinematic-dynamic constrains. What's more, along with the differential flatness property of a quadrotor, the drone is expected to follow the trajectory precisely. So we take states from the trajectory most of the time. 
However, we will take states from the current odometry in some cases in state "GEN_NEW_TRAJ" in ego_replan_fsm.cpp.

closed 2020-09-27 by Bruno-33