ZJU-FAST-Lab/ego-planner #10: How to change the size of the local map？

Robinaoko, 2020-10-01:


bigsuperZZZX, 2020-10-01:
Our planner has no isolated local map, all the map data is stored in the global map (`struct MappingData` in [grid_map.h](https://github.com/ZJU-FAST-Lab/ego-planner/blob/master/src/planner/plan_env/include/plan_env/grid_map.h) ). If the global map is too large for you to allocate, please use 3D Circular Buffer (Vladyslav Usenko, et.al., Real-Time Trajectory Replanning for MAVs using Uniform B-splines and a 3D Circular Buffer).

Robinaoko, 2020-10-01:
Many thanks for your replying！
In the demo，I can see that the uav is planning according to the front sector. Can I change the size of this sector to make it  bigger or smaller?

Robinaoko, 2020-10-01:
I see. I will check the parameters in detail again. Thank you very much!

closed 2020-10-01 by Robinaoko