ZJU-FAST-Lab/ego-planner #11: catkin_make error

shuaiwang888, 2020-10-02:
Hello, I encountered some problems in the compilation process, can you help me to have a look, thank you！
 

In file included from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/src/polynomial_traj.cpp:2:0:
/home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/include/traj_utils/polynomial_traj.h: In member function ‘double PolynomialTraj::getJerk()’:
/home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/include/traj_utils/polynomial_traj.h:237:26: error: no match for ‘operator=’ (operand types are ‘Eigen::internal::enable_if<true, Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double> >::type {aka Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double>}’ and ‘double’)
           mat_jerk(i, j) =
                          ^
In file included from /usr/local/include/eigen3/Eigen/Core:19:0,
                 from /usr/local/include/eigen3/Eigen/Dense:1,
                 from /usr/local/include/eigen3/Eigen/Eigen:1,
                 from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/include/traj_utils/polynomial_traj.h:4,
                 from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/src/polynomial_traj.cpp:2:
/usr/local/include/eigen3/Eigen/src/Core/IndexedView.h:113:3: note: candidate: Eigen::IndexedView<XprType, RowIndices, ColIndices>& Eigen::IndexedView<XprType, RowIndices, ColIndices>::operator=(const Eigen::IndexedView<XprType, RowIndices, ColIndices>&) [with XprType = Eigen::Matrix<double, -1, -1>; RowIndices = double; ColIndices = double]
   EIGEN_INHERIT_ASSIGNMENT_OPERATORS(IndexedView)
   ^
/usr/local/include/eigen3/Eigen/src/Core/IndexedView.h:113:3: note:   no known conversion for argument 1 from ‘double’ to ‘const Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double>&’
In file included from /usr/local/include/eigen3/Eigen/Core:256:0,
                 from /usr/local/include/eigen3/Eigen/Dense:1,
                 from /usr/local/include/eigen3/Eigen/Eigen:1,
                 from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/include/traj_utils/polynomial_traj.h:4,
                 from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/src/polynomial_traj.cpp:2:
/usr/local/include/eigen3/Eigen/src/Core/MatrixBase.h:154:14: note: candidate: template<class OtherDerived> Derived& Eigen::MatrixBase<Derived>::operator=(const Eigen::ReturnByValue<OtherDerived>&) [with OtherDerived = OtherDerived; Derived = Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double>]
     Derived& operator=(const ReturnByValue<OtherDerived>& other);
              ^
/usr/local/include/eigen3/Eigen/src/Core/MatrixBase.h:154:14: note:   template argument deduction/substitution failed:
In file included from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/src/polynomial_traj.cpp:2:0:
/home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/include/traj_utils/polynomial_traj.h:237:26: note:   mismatched types ‘const Eigen::ReturnByValue<Derived>’ and ‘double’
           mat_jerk(i, j) =
                          ^
In file included from /usr/local/include/eigen3/Eigen/Core:256:0,
                 from /usr/local/include/eigen3/Eigen/Dense:1,
                 from /usr/local/include/eigen3/Eigen/Eigen:1,
                 from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/include/traj_utils/polynomial_traj.h:4,
                 from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/src/polynomial_traj.cpp:2:
/usr/local/include/eigen3/Eigen/src/Core/MatrixBase.h:150:14: note: candidate: template<class OtherDerived> Derived& Eigen::MatrixBase<Derived>::operator=(const Eigen::EigenBase<OtherDerived>&) [with OtherDerived = OtherDerived; Derived = Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double>]
     Derived& operator=(const EigenBase<OtherDerived>& other);
              ^
/usr/local/include/eigen3/Eigen/src/Core/MatrixBase.h:150:14: note:   template argument deduction/substitution failed:
In file included from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/src/polynomial_traj.cpp:2:0:
/home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/include/traj_utils/polynomial_traj.h:237:26: note:   mismatched types ‘const Eigen::EigenBase<Derived>’ and ‘double’
           mat_jerk(i, j) =
                          ^
In file included from /usr/local/include/eigen3/Eigen/Core:256:0,
                 from /usr/local/include/eigen3/Eigen/Dense:1,
                 from /usr/local/include/eigen3/Eigen/Eigen:1,
                 from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/include/traj_utils/polynomial_traj.h:4,
                 from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/src/polynomial_traj.cpp:2:
/usr/local/include/eigen3/Eigen/src/Core/MatrixBase.h:146:14: note: candidate: template<class OtherDerived> Derived& Eigen::MatrixBase<Derived>::operator=(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = OtherDerived; Derived = Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double>]
     Derived& operator=(const DenseBase<OtherDerived>& other);
              ^
/usr/local/include/eigen3/Eigen/src/Core/MatrixBase.h:146:14: note:   template argument deduction/substitution failed:
In file included from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/src/polynomial_traj.cpp:2:0:
/home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/include/traj_utils/polynomial_traj.h:237:26: note:   mismatched types ‘const Eigen::DenseBase<Derived>’ and ‘double’
           mat_jerk(i, j) =
                          ^
In file included from /usr/local/include/eigen3/Eigen/Core:265:0,
                 from /usr/local/include/eigen3/Eigen/Dense:1,
                 from /usr/local/include/eigen3/Eigen/Eigen:1,
                 from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/include/traj_utils/polynomial_traj.h:4,
                 from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/src/polynomial_traj.cpp:2:
/usr/local/include/eigen3/Eigen/src/Core/Assign.h:55:30: note: candidate: Derived& Eigen::MatrixBase<Derived>::operator=(const Eigen::MatrixBase<Derived>&) [with Derived = Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double>]
 EIGEN_STRONG_INLINE Derived& MatrixBase<Derived>::operator=(const MatrixBase& other)
                              ^
/usr/local/include/eigen3/Eigen/src/Core/Assign.h:55:30: note:   no known conversion for argument 1 from ‘double’ to ‘const Eigen::MatrixBase<Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double> >&’
In file included from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/include/traj_utils/planning_visualization.h:8:0,
                 from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/src/planning_visualization.cpp:1:
/home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/include/traj_utils/polynomial_traj.h: In member function ‘double PolynomialTraj::getJerk()’:
/home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/include/traj_utils/polynomial_traj.h:237:26: error: no match for ‘operator=’ (operand types are ‘Eigen::internal::enable_if<true, Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double> >::type {aka Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double>}’ and ‘double’)
           mat_jerk(i, j) =
[... 47 lines cut ...]
In file included from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/include/traj_utils/planning_visualization.h:8:0,
                 from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/src/planning_visualization.cpp:1:
/home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/include/traj_utils/polynomial_traj.h:237:26: note:   mismatched types ‘const Eigen::DenseBase<Derived>’ and ‘double’
           mat_jerk(i, j) =
                          ^
In file included from /usr/local/include/eigen3/Eigen/Core:265:0,
                 from /usr/local/include/eigen3/Eigen/Dense:1,
                 from /usr/local/include/eigen3/Eigen/Eigen:1,
                 from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/include/traj_utils/planning_visualization.h:4,
                 from /home/app/ego-planner_ws/src/ego-planner/src/planner/traj_utils/src/planning_visualization.cpp:1:
[... cut ...]
ego-planner/src/planner/traj_utils/CMakeFiles/traj_utils.dir/build.make:86: recipe for target 'ego-planner/src/planner/traj_utils/CMakeFiles/traj_utils.dir/src/polynomial_traj.cpp.o' failed
make[2]: *** [ego-planner/src/planner/traj_utils/CMakeFiles/traj_utils.dir/src/polynomial_traj.cpp.o] Error 1
CMakeFiles/Makefile2:5469: recipe for target 'ego-planner/src/planner/traj_utils/CMakeFiles/traj_utils.dir/all' failed
make[1]: *** [ego-planner/src/planner/traj_utils/CMakeFiles/traj_utils.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

bigsuperZZZX, 2020-10-02:
It's so weird and I tried to fig it out but failed.
It seems that something goes wrong with your Eigen or cpp compiler.

shuaiwang888, 2020-10-02:
Thank you for your answer. I'll try again.

shuaiwang888, 2020-10-02:
It is because I have run lego-loam and sc-lego-loam. There is a conflict between the eigen versions. After I uninstall eigen3, the problem is solved. Thank you for your advice and wish you a happy holiday!

closed 2020-10-02 by shuaiwang888