ZJU-FAST-Lab/ego-planner #13: How to display all the esdf maps during the flight like video and indoor.gif?

Robinaoko, 2020-10-26:
When I run the demo, I can only dispaly the esdf map around the UAV. How can I display all the esdf maps during the flight like video and indoor.gif?

bigsuperZZZX, 2020-10-26:
First, the map you see is grid map, not esdf map.
If you want to display the whole map,  you can enlarge the parameter "grid_map/local_map_margin" in [advanced_param.xml](https://github.com/ZJU-FAST-Lab/ego-planner/blob/master/src/planner/plan_manage/launch/advanced_param.xml), which infers the grid size to be displayed.
But I have to remind you that map display requires a lot of computation, so we only record a small local map during the flight and wrote a new node to display it Incrementally offline.

Robinaoko, 2020-10-26:
OK,many thanks for your help!

closed 2020-10-26 by Robinaoko

1997-X, 2023-02-21:
Can you explain in detail how the map shows your realization?
Look forward to your reply！！！！

bigsuperZZZX, 2023-03-16:
Maybe this callback function is what you want: 
https://github.com/ZJU-FAST-Lab/ego-planner/blob/fcdfc5419dcc0c93105803e6f60b1b08b4e5152d/src/planner/plan_env/src/grid_map.cpp#L646