ZJU-FAST-Lab/ego-planner #16: find bug on [ ego-planner/src/planner/bspline_opt/src/bspline_optimizer.cpp ]

FranklinBF, 2021-02-19:
Line 80:
should be :

else if(occ && last_occ)
{
          ++same_occ_state_times;
 }

FranklinBF, 2021-02-19:
I meet some bug sometimes when running this project, and I find may be this line is wrong.

bigsuperZZZX, 2021-03-07:
Sorry for replying so late, since I have been working on a paper submission recently.
Could you describe how the bug looks like?
I have exam the code but can't fig out the negative affect of the current version.

bigsuperZZZX, 2021-04-04:
Hello, if you have no other questions, please close this issue.

FranklinBF, 2021-04-04:
Sorry for late replying. 
I'm not sure if it will cause an error,  just find sometimes it is not stable, and I changed it  and it then has less error.

Also, the logic here, I think is to take "else if(occ && last_occ)" not just "else if(occ)“. This is what I think. I don't know if it is correct from your point of view.

I also want to thank for your work! I have use your ego planner for 2D trajectory planning in dynamic environment for my master thesis, which shows good performance due to fast replanning and not need for a collision-free initial guess.

bigsuperZZZX, 2021-04-04:
I examed the code again. If the code is modified as you said, [this line](https://github.com/ZJU-FAST-Lab/ego-planner/blob/f533ecbd2978b69e13a75d082305d769d79c5443/src/planner/bspline_opt/src/bspline_optimizer.cpp#L66) is true only if "i==order_". That can cause a negative effect since the later points with "i>order_" will never take as occuped in this function.

FranklinBF, 2021-04-04:
However, if you modified as I said
same_occ_state_times will still counting the points that continuously in occupany.
         
The following condition can be  met.
same_occ_state_times > ENOUGH_INTERVAL 

Also, another question, why you are searching here for points with continuous same states, otherthan points with continuous occupied states?

bigsuperZZZX, 2021-04-06:
For the modified code, since [this line](https://github.com/ZJU-FAST-Lab/ego-planner/blob/f533ecbd2978b69e13a75d082305d769d79c5443/src/planner/bspline_opt/src/bspline_optimizer.cpp#L66) is always false except when "i==order_", "[flag_got_start](https://github.com/ZJU-FAST-Lab/ego-planner/blob/f533ecbd2978b69e13a75d082305d769d79c5443/src/planner/bspline_opt/src/bspline_optimizer.cpp#L69)" will not get true. Then it results in the occupied segment be ignored at [this line](https://github.com/ZJU-FAST-Lab/ego-planner/blob/f533ecbd2978b69e13a75d082305d769d79c5443/src/planner/bspline_opt/src/bspline_optimizer.cpp#L93).

For the question: the reason for this trick is to make sure each segment has enough length and two adjacent segments have enough interval.

FranklinBF, 2021-04-09:
I see. I thought it is used to find the first grid that meets  a collision. Thank you very much for the detailed explanation.
By the way, I meet another challenge using EGO planner in dynamic environment:
If start_pos or first 3 points is in collision, then EGO planner cannot execute the optimization.
Is there any solution to this problem?

bigsuperZZZX, 2021-04-13:
It should not happen. This error means the drone is in obstacles now, which indicates a collision in the realworld.
Note that we use an inflated map for planning. So please subscribe the inflated map in rviz and check why this happens.

FranklinBF, 2021-05-10:
Thank you so much for the explanation and the issue is solved.

The problem I meant was not in the EGO-Planner project (EGO planner runs perfectly),  but in my own project, where I have modified your EGO planner method for a diff-drive robot planning in an environment with crowds.

closed 2021-08-05 by USTfgaoaa

zrd1234, 2022-09-23:
> 很抱歉迟到。我不确定它是否会导致错误，只是发现有时它不稳定，我改变了它，然后它的错误更少。
> 
> 另外，这里的逻辑，我认为是采取“否则如果（occ &last_occ）”而不仅仅是“否则如果（occ）”。这就是我的想法。我不知道从你的角度来看是否正确。
> 
> 我还要感谢你的工作！我的硕士论文中使用了您的自我规划器在动态环境中进行2D轨迹规划，由于快速重新规划并且不需要无碰撞的初始猜测，因此表现出良好的性能。
Hello, I see that you use 2D for planning, and now I also want to use 2D, may I ask how to modify the code? Thank you for your guidance