ZJU-FAST-Lab/ego-planner #21: Question about minSnapTraj function

dssdyx, 2021-05-27:
In line 78 of polynomial_traj.cpp,  the number of fixed derivative is 2*seg_num + 4 explained as 3 +3 +(seg_num -1 )*2 = 2*seg_num +4. But, I think the dF should be p,v,a of head and end , and p of middle points. It's actually 3 + 3 + seg_num -1 = seg_num + 7. I feel confused about this. Could you help me answer my question? Thanks!

bigsuperZZZX, 2021-05-27:
The Selection Matrix "C" has to map a middle point to both 2 segments just before and after that middle point, therefore it multiples the number of middle points by 2.
For more details, please refer to the paper "Polynomial Trajectory Planning for Aggressive Quadrotor Flight in Dense Indoor Environments" in Section "3 Polynomial Trajectory Optimization".

dssdyx, 2021-05-28:
> The Selection Matrix "C" has to map a middle point to both 2 segments just before and after that middle point, therefore it multiples the number of middle points by 2.
> For more details, please refer [...]

I have read that paper and I think maybe it's just the different form of same thing. The 2seg_num+4 dF and seg_num +7 dF are same. They have their own Ct to transform them to d_all which is [p0 v0 a0 p1 v1 a1, p1 v1 a1 p2 v2 a2,...] .
Thanks for your answer!

closed 2021-05-28 by dssdyx