ZJU-FAST-Lab/ego-planner #25: Improper camere-IMU time synchronization on Realsense D435i with the driver modifications

Pallav1299, 2021-09-01:
I tried using the modified realsense-ros driver to get input for Stereo-inertial SLAM using VINS-Fusion. 

Although the projector dots were almost removed, I witnessed that the changes affected the IMU and camera timestamps. 
- How can this affect the SLAM performance?
- How should we ideally use it for Stereo-inertial SLAM configuration?
- I also noticed that on using the modified version of realsense-ros driver with VINS-fusion promped "wait for imu ..." several times. Does it  mean that the changes affected the time synchronization between camera and IMU?

Open to suggestions.

bigsuperZZZX, 2021-09-06:
I think this function will not affect the SLAM performance since we have been using it for a long time.
You can adjust the frame rate in launch files so that the camera outputs at the same rate as not using such a feature.
One minor difference is the time stamps between depth and gray-scale images since they are outputted alternatively.
But this difference should not affect the mapping performance since the time deviation is very small (~16ms at 30fps setting).

bigsuperZZZX, 2021-09-17:
It no more questions, I will close this issue.

closed 2021-09-17 by bigsuperZZZX

Pallav1299, 2021-09-24:
@bigsuperZZZX, Thanks for the clarification.

Pallav1299, 2021-10-05:
I noticed a difference in IMU and Camera timestamps. Please check the attached images for the same. I've faced this issue earlier as well. Setting the "initial_reset" flag to "true" acted as a workaround. But not this time. 

This issue leads to degradation in SLAM preformance in VINS.

**Camera timestamp:**
[image]

**IMU timestamp:**
[image]