ZJU-FAST-Lab/ego-planner #26: On which topic position and velocity commands for tajactory is published?

shubham-shahh, 2021-09-10:
Hi,
I am trying to merge the planner with Iris copter in the gazebo with intel realsense D435, I am passing the iris odom data and pont cloud data to ego planner, I am using the /planning/pos_cmd topic  data to move the Iris copter but, /planning/pos_cmd publishes final goal and the copter reaches the goal without following the copter.

Which topic shall I listen to, to move the Iris copter based on the trajactory generated? @bigsuperZZZX  @USTfgaoaa

bigsuperZZZX, 2021-09-10:
/planning/pos_cmd is the right topic for these commands, there must be something wrong.
Please subscribe to various visualization topics in rviz to find out the mistake.
Also, you can compare the outputs between the original code downloaded from this repo and your modified code to discover unnormal behaviors.

shubham-shahh, 2021-09-10:
Hi @bigsuperZZZX  thanks for the response, I am using [this](https://github.com/HKPolyU-UAV/E2ES) repo for simulation. 

so ideally, /planning/pos_cmd should publish the trajectory points sequentially, right?

bigsuperZZZX, 2021-09-10:
That's right.

shubham-shahh, 2021-09-10:
since I am using gazebo simulator, how can I stop S03 simulator?

bigsuperZZZX, 2021-09-10:
Just comment that node in https://github.com/ZJU-FAST-Lab/ego-planner/blob/master/src/planner/plan_manage/launch/simulator.xml

shubham-shahh, 2021-09-10:
> Just comment that node in https://github.com/ZJU-FAST-Lab/ego-planner/blob/master/src/planner/plan_manage/launch/simulator.xml

shall I comment so3_quadrotor_simulator and nodelet, both the nodes?

bigsuperZZZX, 2021-09-10:
Both nodes

shubham-shahh, 2021-09-10:
> Both nodes

okay, thank you so much for the help, I'll try it

shubham-shahh, 2021-09-10:
> Both nodes

I am publishing  the /planning/poa_cmd data on /mavros/setpoint_raw/local topic, is the that correct mavros topic?

shubham-shahh, 2021-09-10:
> Hi @bigsuperZZZX thanks for the response, I am using [this](https://github.com/HKPolyU-UAV/E2ES) repo for simulation.
> 
> so ideally, /planning/pos_cmd should publish the trajectory points sequentially, right?

What are the major changes to be made to adopt ego planner to https://github.com/HKPolyU-UAV/E2ES this gazebo simulation

bigsuperZZZX, 2021-09-10:
> > Both nodes
> 
> I am publishing the /planning/poa_cmd data on /mavros/setpoint_raw/local topic, is the that correct mavros topic?

They are incompatible.

I am not familiar with https://github.com/HKPolyU-UAV/E2ES, but that's true that the /planning/pos_cmd topic is the output of the planner to some controllers, so you should find some way or write some code by yourselves to make your drone receive, recognize and execute the /planning/pos_cmd.

shubham-shahh, 2021-09-10:
> They are incompatible.
>
> I am not familiar with https://github.com/HKPolyU-UAV/E2ES, but that's true that the /planning/pos_cmd topic is the output of the planner to some controllers, so you should [...]

Okay thanks for all the help

closed 2021-09-15 by shubham-shahh