ZJU-FAST-Lab/ego-planner #33: How to get a simulation picture in your paper:EGO-Planner: An ESDF-free Gradient-based Local Planner for Quadrotors

Long-Wu-code, 2021-12-10:
Hi：
      I want to konw how to get a simulation picture form your code,  just like this in your paper:
              https://user-images.githubusercontent.com/63492361/145538720-4f84d456-9387-40a4-b495-21dd7670ee12.png
      and this:
             https://user-images.githubusercontent.com/63492361/145539035-d12db0f3-fd7c-44c4-95a4-3e0772c9eec7.png
       While  running  the simulation, it only show part of mapping ,like this:
            https://user-images.githubusercontent.com/63492361/145539723-88920ceb-f8c3-4235-9615-d7e6cc7e047c.png
 
       Look forward to your favourable reply

bigsuperZZZX, 2021-12-10:
Just increasing the "Decay Time" in the pointcloud2 topic in the Displays panel.
It's the parameter telling rviz how much time to keep a pointcloud message.

Long-Wu-code, 2021-12-10:
> Just increasing the "Decay Time" in the pointcloud2 topic in the Displays panel. It's the parameter telling rviz how much time to keep a pointcloud message.
Thank you for your reply, but it seems doesn't work:
https://user-images.githubusercontent.com/63492361/145570689-eb7c4764-f761-481d-939f-5f233c3e58f4.png
It's there something else I need to set?

bigsuperZZZX, 2021-12-14:
It seems that you are editing the wrong topic.
Try to uncheck each topic and see which one is currently displaying.

bigsuperZZZX, 2022-01-14:
I'll close this issue if there are no more questions.

closed 2022-01-14 by bigsuperZZZX

Long-Wu-code, 2022-02-16:
> It seems that you are editing the wrong topic. Try to uncheck each topic and see which one is currently displaying.
     I'm sorry to reply you now , because something else I stopped this work a few month ago! 
     Today, I follow you advice try every topic in PointCloud2（eg:/grid_map/occupancy, /grid_map/occupancy_inflate,/grid_map/unknown,/map_generator/global_cloud,/pcl_render_node/cloud）, but still didn't work.
Hope for your reply, thank you so much