ZJU-FAST-Lab/ego-planner #35: Some error appear, after setting'4.Use GPU or Not‘

WjMessi1, 2022-02-09:


bigsuperZZZX, 2022-02-12:
All Cuda problems we encountered until now are all relevant to corrupted Cuda installation. 
Therefore we recommend you check your installation of Cuda or try the planner on another computer.
Hope this works.

WjMessi1, 2022-02-12:
> All Cuda problems we encountered until now are all relevant to corrupted Cuda installation. Therefore we recommend you check your installation of Cuda or try the planner on another computer. Hope this works.

ok， thanks for your reply！ I am trying it on a new computer

WjMessi1, 2022-02-14:
I have reinstalled my system. Configuration is ubuntu20.04, cuda11.6, now, the result of order `nvidia-smi` as follow:
[image]
I can pass `catkin_make`, but still there are some warning：
[image]
when I run `roslaunch ego_planner simple_run.launch`，no error occured, but some warning like this:
[image]
I think it's related to the warning occur during `catkin_make`. Then the deep images are all black, the output of topic `/pcl_render_node/depth` are all 0.
[image]
Can this problem be solved?
One more question is how should I get the speed of the drone in the simulation? By which topic to echo? Thanks!!!

bigsuperZZZX, 2022-02-14:
I also use Cuda 11 but no error occurs during compiling.
The warning during code running is relevant to rviz visualization and is nothing to do with Cuda, so you can just ignore it or convert the initial _.mesh_ file to a _.dae_ file to eliminate the warning.

Remember to config this:
[image].
On my computer installed with Cuda 11 + 1660Ti, I set this to "-gencode arch=compute_75,code=sm_75" to enable depth rendering.

bigsuperZZZX, 2022-02-14:
Sorry, I accidentally closed this issue. It has been reopened again.

WjMessi1, 2022-02-14:
> Sorry, I accidentally closed this issue. It has been reopened again.

ok, thanks for your reply, I have tried your recommand method above, this still has no effect...... I think this warning has a great impact on planning. As you can see:
[image]
The trajectory goes straight through the obstacle, almost in a straight line and no depth image output. I've had success with GPU before, but recently don't know what happened... I'm sorry, this question took your time. If it doesn't work, I will try to reinstall the system again. It may be the cause of cuda.
At last，how can I get the speed of the drone during operation in the simulation? By which rostopic？

WjMessi1, 2022-02-17:
> Yes if there are no valid depth images, the planner has no obstacle information.
> 
> To get the drone speed, you just listen to the Odometry topic, which may be, reads like "/visual_slam/odom".

Hello! This problem is solved, it is really metaphysics. I reinstalled the system again, now the version of cuda is 11.2, and the nvidia-driver is 450, the previous problem may be the version of the kernel of the lunix system. Thank you！ I think you can close it this problem.

WjMessi1, 2022-02-17:
How should I modify the detection range of the depth camera? The default detection range parameter seems to have no limit.

bigsuperZZZX, 2022-03-05:
Our code doesn't provide this interface, but you can modify it just in the depth image callback function where you can check all the pixels and manually set the value which exceeds your threshold to zero.

bigsuperZZZX, 2022-03-25:
If there are no more questions, I will close this issue.

closed 2022-03-25 by bigsuperZZZX

WjMessi1, 2022-03-27:
> 

ok， thank you！