ZJU-FAST-Lab/ego-planner #38: Questions about airsim simulation

rsj007, 2022-03-24:
Hello, I am running ego-planner simulation in airsim.
I use depth and camera pose and set the tfs as follows. However the grid map is strange, only a small one, not like the tree before the drone. Besides when the drone take offs, the grid map disappears. 

<node pkg="tf" type="static_transform_publisher" name="odom_to_front_center_optical" args="0 0 0 0 0 0 odom front_center_optical 30" />
<node pkg="tf" type="static_transform_publisher" name="odom_to_world" args="0 0 0 0 0 1.57 odom world 100" />

[image]

bigsuperZZZX, 2022-03-25:
Obviously, the extrinsic parameters of the camera are inappropriate. 
Check the framework of your camera pose topic, we don't know what frame airsim is used.
Then refer to [this line of code](https://github.com/ZJU-FAST-Lab/ego-planner/blob/fcdfc5419dcc0c93105803e6f60b1b08b4e5152d/src/planner/plan_env/src/grid_map.cpp#L998) to write your own code for rotating your acquired camera pose.

rsj007, 2022-03-29:
I only set the tf as below:
` <node pkg="tf" type="static_transform_publisher" name="camera_link_to_world" args="0 0 0 -1.57 0 -1.57  base_link front_center_optical 30" />
    <node pkg="tf" type="static_transform_publisher" name="tf_camera" args="0 0 0.0 0 0 0 base_link odom 100"/>`

Is it right? Does I need other tfs?   Thanks very much!

LvAoAo, 2022-04-11:
> I use the point cloud as input. The tfs I set are as below:
> ` <node pkg="tf" type="static_transform_publisher" name="camera_link_to_world" args="0 0 0 -1.57 0 -1.57 base_link front_center_optical 30" /> <node pkg="tf" type="static_transform_publisher" name="tf_camera" args="0 0 0.0 0 0 0 base_link odom 100"/>`

Hello! I may meet the same question...
So could you please share something about how to get the correct depth image required by the "ego-planner"?I input the depthplanar and just get  2-value images just like the photo.....And I nearly get no grid map any time.Only when the drone is encountered with an obstacle can i got a slight gridmap...
[image]

bigsuperZZZX, 2022-04-25:
Our mapping code does not use tf to get extrinsic parameters.
You should manually set it directly in the code as I mentioned in the previous response.

bigsuperZZZX, 2022-07-10:
If there is no more reply, I will close this issue

closed 2022-07-10 by bigsuperZZZX