ZJU-FAST-Lab/ego-planner #40: The reason to suppress the slope of the collisioin cost

Joeyyuenjoyslife, 2022-04-10:
Hi, Here is Joey. I want to know is there any specific reason to suppress the slope of the collision cost item when distance is negative? i can't find further explanation in the paper, thanks a lot !
[image]

bigsuperZZZX, 2022-04-25:
Well, it's a little bit complicated. We use a quasi-Newton method L-BFGS as our optimizer described in section "B. Numerical Optimization". It approximates the optimized function in a quadratic form. Therefore the more similar we formulate the cost function to a quadratic form, the faster convergence we can achieve in theory.

Joeyyuenjoyslife, 2022-04-25:
> Thanks for your reply! it makes sense, I will close this issue.

closed 2022-04-25 by Joeyyuenjoyslife