ZJU-FAST-Lab/ego-planner #43: some question about uav_simulator

Mealoore, 2022-07-06:
When using your so3 simulator, I publish the command
```
header: 
  seq: 21
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
position: 
  x: nan
  y: nan
  z: nan
velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
acceleration: 
  x: 0.0
  y: 0.0
  z: 0.0
yaw: 2.0
yaw_dot: 0.0
kx: [0.0, 0.0, 0.0]
kv: [0.0, 0.0, 0.0]
trajectory_id: 0
trajectory_flag: 0
```
the height of uav is not correct, I think it should be at original point

bigsuperZZZX, 2022-07-10:
I'm not quite sure about that. You could compare the original messages from the traj_server node with your messages to find the problem.

Mealoore, 2022-07-10:
> 
As for my question, that is to say, how should I pub the message, for speed control and yaw or yaw rate control?

bigsuperZZZX, 2022-07-10:
The simulator does not support yaw rate control. 
As for other control methods, please check [this function](https://github.com/ZJU-FAST-Lab/ego-planner/blob/fcdfc5419dcc0c93105803e6f60b1b08b4e5152d/src/uav_simulator/so3_control/src/SO3Control.cpp#L38) for the details. There are too many flags and code logics that I can't remember precisely.

Mealoore, 2022-07-10:
> The simulator does not support yaw rate control. As for other control methods, please check [this [...]

how about yaw control? I just need speed control and  yaw control without position control.

bigsuperZZZX, 2022-07-11:
Please check that function of dozens of lines to find your solution.

closed 2022-08-23 by bigsuperZZZX