ZJU-FAST-Lab/ego-planner #48: How to remove obstacles in the model

zrd1234, 2022-08-29:
Hello, how should I modify to remove the obstacles in the code? I want to observe the real world with my camera and display it in RVIZ. Thank you very much.

bigsuperZZZX, 2022-08-29:
Just comment this block https://github.com/ZJU-FAST-Lab/ego-planner/blob/fcdfc5419dcc0c93105803e6f60b1b08b4e5152d/src/planner/plan_manage/launch/simulator.xml#L45

zrd1234, 2022-08-29:
[image]

Thanks for your reply, I have blocked all the codes here, and the operation effect is shown in the picture. The camera is D435i, and the algorithm is Vins-Mono. I want to get the real environment, and as my device moves, it also moves, please help me, thank you.

zrd1234, 2022-09-06:
> Just comment this block
> 
> https://github.com/ZJU-FAST-Lab/ego-planner/blob/fcdfc5419dcc0c93105803e6f60b1b08b4e5152d/src/planner/plan_manage/launch/simulator.xml#L45

Hello, thank you for your reply. I have a new problem now.
After I passed in the topic of Vins-Mono, the picture was always on the right plane of the camera. May I ask how to solve this problem?

closed 2022-09-13 by zrd1234