ZJU-FAST-Lab/ego-planner #51: Real environment trajectory generation

zrd1234, 2022-09-13:
Hello, I have imported the real environment and want to make planning, but after I set the target point, it automatically generates a track to reach the target point. How can I modify it? After I select the target point, the track is generated with the movement of my camera

closed 2022-09-16 by zrd1234

TanmayJain03, 2023-03-16:
@zrd1234 could you solve it, I too am facing the same issue where the start of the planned trajectory moves ahead while the drone is still stationary, and the tracker suddenly starts moving aggressively in a straight line directly towards the already leading trajectory's starting point till it catches up also not avoiding any obstacles during this.
I am trying to make changes in ego_planner_fsm.cpp where I changed the start_pt in the planFromCurrentTraj() function to the position of the drone as observed in gazebo model states.
Could you please share any lead as to how you tried to fix the problem, any help would be really appreciated.
Regards

bigsuperZZZX, 2023-03-18:
https://github.com/ZJU-FAST-Lab/ego-planner/issues/14#issuecomment-1472799232