ZJU-FAST-Lab/ego-planner #52: 3D-2D

zrd1234, 2022-09-16:
Hello, thank you very much for your work. I want to use it in 2D to make a plane plan and how to modify it. Please guide me. Thank you.

bigsuperZZZX, 2022-10-08:
You can just inflate the gird map a great amount in z direction.

zrd1234, 2022-10-08:
> You can just inflate the gird map a great amount in z direction.

Hello, is the z-axis in the configuration file? I have modified the A* algorithm to two-dimensional search, but sometimes there is A star error problem, may I ask what is the reason for this problem?
[image]

zrd1234, 2022-10-17:
Hello, I fixed the search range of A* in the plane, so it is a 2D path planning, but a star error often occurs. If the a* algorithm fails, the path becomes 3 dimensional. Would like to ask the most likely cause of failure.

bigsuperZZZX, 2023-01-29:
I hope that this late response helps. Your error indicates that the [openSet_](https://github.com/ZJU-FAST-Lab/ego-planner/blob/c50669a58ae51e6e2a9c35dc614b153b9b4c8c75/src/planner/path_searching/src/dyn_a_star.cpp#L162) is empty before finding an A* solution.

zrd1234, 2023-01-29:
> I hope that this late response helps. Your error indicates that the [openSet_](https://github.com/ZJU-FAST-Lab/ego-planner/blob/c50669a58ae51e6e2a9c35dc614b153b9b4c8c75/src/planner/path_searching/src/dyn_a_star.cpp#L162) is empty before finding an A* solution.
Thank you very much for your reply. Do you mean that this error occurred because openset_ was empty and the node could not be found?

bigsuperZZZX, 2023-01-29:
That's true

zrd1234, 2023-01-29:
> 这是真的

May I ask if there is any good way to solve this problem?
I want to do two-dimensional planning, but sometimes I just go three-dimensional.

bigsuperZZZX, 2023-01-29:
The planner is designed for 3D flight, and maybe converting to 2D will be a big project.
A relatively easier solution is to modify [the inflation function](https://github.com/ZJU-FAST-Lab/ego-planner/blob/fcdfc5419dcc0c93105803e6f60b1b08b4e5152d/src/planner/plan_env/src/grid_map.cpp#L521) in the mapping code.
You can choose a certain layer at a specific height and inflate that layer up and down with enough margin so that the trajectory will not go above the wall. It will make the new 3D map looks like this:
[image]
Then implement the planner as normal and ignore the z-component of the trajectory.

zrd1234, 2023-01-29:
> 
May I ask how to operate or modify it? I don't understand this part of the code.
Because of my limited ability, I did not see how to operate.

zrd1234, 2023-01-29:
> The planner is designed for 3D flight, and maybe converting to 2D will be a big project. A relatively easier solution is to modify [the inflation [...]

May I ask how to operate or modify it? I don't understand this part of the code.
Because of my limited ability, I did not see how to operate.

bigsuperZZZX, 2023-01-29:
modifying and debugging this is still a little bit complicated. So perhaps it is unavoidable for you to read and understand the code by yourself.

closed 2023-03-31 by zrd1234