ZJU-FAST-Lab/ego-planner #53: Modify Track Direction

zrd1234, 2022-10-10:
Hello, I recently applied the sub algorithm to the quadruped, but found that the quadruped's motion path and trajectory generation direction are mirrored. How can I modify the coordinate system so that the trajectory coordinates correspond to the quadruped.

bigsuperZZZX, 2022-10-17:
Please elaborate on the details of the problem

zrd1234, 2022-10-17:
> Please elaborate on the details of the problem

Hello, the problem should be the axis. However, the initialization direction of vins-mono is the Y-axis, resulting in a 90-degree deviation. I changed the initialization direction of the map to the Y-axis. Now, every time the path is planned, the robot will rotate 90 degrees and then move according to the planned speed and angular velocity, that is, the planning is to the right front. But the robot will rotate 90 degrees and move toward the front right of the rotation. I want to know how to modify the camera coordinate system obtained by ego.
In addition, I would like to ask why a star error often occurs. thank you

bigsuperZZZX, 2022-10-17:
1. Camera coordinate system can be modified through [this line](https://github.com/ZJU-FAST-Lab/ego-planner/blob/fcdfc5419dcc0c93105803e6f60b1b08b4e5152d/src/planner/plan_env/src/grid_map.cpp#L91).
2. I have no idea why the robot rotates 90 degrees. It seems a controller problem.
3. A star error occurs for many reasons. It doesn't matter if the planner works well.

closed 2022-10-17 by zrd1234