ZJU-FAST-Lab/ego-planner #54: Modify Track Control Direction

zrd1234, 2022-10-20:
Hello, I'm sorry to bother you again.
When Realsense is used, the default camera front is facing the Y axis after the VINS initialization; The track command output by Ego Planner always faces the X axis at the initial time. I would like to ask you how to increase the angle at the initial time so that it can issue commands on the y-axis. thank you.

bigsuperZZZX, 2022-10-28:
"When Realsense is used, the default camera front is facing the Y axis after the VINS initialization": this is not normal. The camera should face the X axis by default.

zrd1234, 2022-10-28:
> “使用实感时，VINS初始化后默认摄像头正面朝向Y轴”：这是不正常的。默认情况下，相机应面向 X 轴。

hello
The initialization of vins-mono is on the Y-axis, because the realsense camera's outward direction is on the Y-axis and should not be initially on the X-axis.

closed 2023-01-29 by bigsuperZZZX