ZJU-FAST-Lab/ego-planner #59: Changing quadrotor frame class and integration on custom worlds

IllierZer, 2022-12-08:
First of all thanks a lot for releasing the ego-planner as open-source, this is really useful. I had some doubts regarding the custom implementation of this package. Firstly, I would like to know how I can change the frame class of the quadrotor (currently the frame class is '+' (plus configuration), I want to simulate an 'x' (cross configuration) quadrotor), please provide details on how to do so. Secondly, The map generator in the simulation.xml seems to be one way to change the world in which the planner works. For example, what exactly am I supposed to do if I want to use the planner on a custom gazebo world?

bigsuperZZZX, 2023-01-29:
1. The frame class has nothing to do with the planner. You can set [this](https://github.com/ZJU-FAST-Lab/ego-planner/blob/fcdfc5419dcc0c93105803e6f60b1b08b4e5152d/src/uav_simulator/Utils/odom_visualization/src/odom_visualization.cpp#L448) to true for cross configuration.
2. All you have to do is provide a depth stream on [this topic](https://github.com/ZJU-FAST-Lab/ego-planner/blob/fcdfc5419dcc0c93105803e6f60b1b08b4e5152d/src/planner/plan_manage/launch/run_in_sim.launch#L22) with corresponding [camera pose](https://github.com/ZJU-FAST-Lab/ego-planner/blob/fcdfc5419dcc0c93105803e6f60b1b08b4e5152d/src/planner/plan_manage/launch/run_in_sim.launch#L21).

IllierZer, 2023-02-01:
> 1. The frame class has nothing to do with the planner. You can set [...]

Thanks a lot for your reply. Another set of challenges that I have been facing are :
1. What exactly is the output of the planner (is it a trajectory?)
2. Whatever be the output of the planner, how do I integrate the control commands I get from the planner with ArduPilot. Basically the control commands that are supposed to send the planner command to the drone using ArduPilot.

bigsuperZZZX, 2023-02-08:
1. A trajectory that is published to the node "traj_server".
2. If you want to fly a real drone, please follow this [detailed tutorial](https://github.com/ZJU-FAST-Lab/Fast-Drone-250). However, we use PX4 instead of ArduPilot.