ZJU-FAST-Lab/ego-planner #60: 编译ego_planner时报错/an error when compiling ego_planner

ADgai234, 2023-01-08:
报错如下/The following are errors

Errors     << ego_planner:cmake /home/lad/catkin_ws/logs/ego_planner/build.cmake.002.log
CMake Warning (dev) at CMakeLists.txt:2 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.
 
  The following variable(s) would be set to empty:
 
    CMAKE_PROJECT_VERSION
    CMAKE_PROJECT_VERSION_MAJOR
    CMAKE_PROJECT_VERSION_MINOR
    CMAKE_PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.
 
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "quadrotor_msgs"
  with any of the following names:
 
    quadrotor_msgsConfig.cmake
    quadrotor_msgs-config.cmake
 
  Add the installation prefix of "quadrotor_msgs" to CMAKE_PREFIX_PATH or set
  "quadrotor_msgs_DIR" to a directory containing one of the above files.  If
  "quadrotor_msgs" provides a separate development package or SDK, be sure it
  has been installed.

quadrotor_msgs这个包我有下载，是不是要添加什么路径（Add the installation prefix of "quadrotor_msgs" to CMAKE_PREFIX_PATH ），我是小白，试了很多方法都不行，可不可以麻烦帮我看看
I have downloaded the package"quadrotor_msgs".I think maybe I need to solve the problems about path（Add the installation prefix of "quadrotor_msgs" to CMAKE_PREFIX_PATH ）.I have tried many ways.However,they didn't work.Could you please help me?

70loKirin, 2023-01-08:
Ubuntu20.04的话`sudo apt install ros-noetice-quadrotor-msgs`即可
其他版本需替换noetice字段

ADgai234, 2023-01-09:
> Ubuntu20.04的话`sudo apt install ros-noetice-quadrotor-msgs`即可 其他版本需替换noetice字段

这个我试过，应该是sudo apt install ros-noetic-quadrotor-msgs吧，但是说定位不到软件包

bigsuperZZZX, 2023-01-29:
这软件包是写在程序里的，apt装不了的，这个目录下：src/uav_simulator/Utils/quadrotor_msgs
你可以先编译一下这个软件包，然后再编译全部

ADgai234, 2023-01-29:
> 

谢谢！我之前就是这么操作的，然后成功了

ADgai234, 2023-01-29:
> 这软件包是写在程序里的，apt装不了的，这个目录下：src/uav_simulator/Utils/quadrotor_msgs 你可以先编译一下这个软件包，然后再编译全部

谢谢！我之前就是这么操作的，然后成功了

closed 2023-01-29 by bigsuperZZZX