ZJU-FAST-Lab/ego-planner #62: 启动ego_planner报错

ADgai234, 2023-01-15:
ubuntu18.04 Melodic
执行以下命令/ran the command
`roslaunch ego_planner single_uav.launch `

报错如下/error
`[ WARN] [1673777077.594699933, 1980.133000000]: Waiting for trigger from [n3ctrl] from RC
[iris_0_ego_planner_node-3] process has died [pid 26423, exit code -9, cmd /home/lad/catkin_ws/devel/lib/ego_planner/ego_planner_node ~odom_world:=/vins_estimator/odometry ~planning/bspline:=/xtdrone/iris_0/planning/bspline ~planning/data_display:=/xtdrone/iris_0/planning/data_display ~planning/broadcast_bspline_from_planner:=/broadcast_bspline ~planning/broadcast_bspline_to_planner:=/broadcast_bspline ~grid_map/odom:=/xtdrone/iris_0/vins_estimator/odometry ~grid_map/cloud:=/iris_0/pcl_render_node/points ~grid_map/pose:=/iris_0/camera_pose ~grid_map/depth:=/iris_0/realsense/depth_camera/depth/image_raw __name:=iris_0_ego_planner_node __log:=/home/lad/.ros/log/8e05f104-94bb-11ed-b8dc-08002717c910/iris_0_ego_planner_node-3.log].
log file: /home/lad/.ros/log/8e05f104-94bb-11ed-b8dc-08002717c910/iris_0_ego_planner_node-3*.log
`
请问该怎么解决

bigsuperZZZX, 2023-01-29:
这个报错没有提供任何信息。你可以看看最后一行“log file: /home/lad/.ros/log/8e05f104-94bb-11ed-b8dc-08002717c910/iris_0_ego_planner_node-3*.log”这个日志文件的内容

ADgai234, 2023-01-29:
> 这个报错没有提供任何信息。你可以看看最后一行“log file: /home/lad/.ros/log/8e05f104-94bb-11ed-b8dc-08002717c910/iris_0_ego_planner_node-3*.log”这个日志文件的内容

好的谢谢！

Image-cong, 2023-03-10:
请问为什么找得到相应报错日志文件的文件夹（里面有master.log/rosoutl.log等），ls -a也试过，但是没有找到iris_0_ego_planner_node-3*.log文件，求助。

Bruce-Si, 2023-03-14:
> 请问为什么找得到相应报错日志文件的文件夹（里面有master.log/rosoutl.log等），ls -a也试过，但是没有找到iris_0_ego_planner_node-3*.log文件，求助。

同找不到log文件...

bigsuperZZZX, 2023-03-16:
magic.
根据我的经验，建议检查检查是不是当前定位位置跑到你定义的地图范围的外面去了

tianyuZ, 2023-03-23:
> ubuntu18.04 Melodic 执行以下命令/ran the command `roslaunch ego_planner single_uav.launch `
>
> 报错如下/error `[ WARN] [1673777077.594699933, 1980.133000000]: Waiting for trigger from [n3ctrl] from RC [...]

我也遇到了相同问题，请问你已经解决了吗？

bigsuperZZZX, 2023-03-24:
> magic. 根据我的经验，建议检查检查是不是当前定位位置跑到你定义的地图范围的外面去了

先按我说的测试

cheslee-z, 2023-05-20:
我之前用catkin_build编译也遇到了这个问题，后来我换用catkin make编译就好了

bigsuperZZZX, 2023-06-29:
默认是得用catkin make编译的，catkin_build没测试过

closed 2023-06-29 by bigsuperZZZX

wdczz, 2023-07-29:
> 我之前用catkin_build编译也遇到了这个问题，后来我换用catkin make编译就好了

我认为这是玄学，但是我先试试TUT

wdczz, 2023-07-29:
> 默认是得用catkin make编译的，catkin_build没测试过

测试了，问题依旧，我认为这不是问题的关键TUT

wdczz, 2023-07-29:
确实可以运行了，但是很快就挂了
hit: 0.619039
miss: -0.619039
min log: -1.99243
max: 2.19722
thresh log: 1.38629
[ INFO] [1690624557.918143085, 1999.837000000]: Wait for 1 second.
[FSM]: from INIT to WAIT_TARGET
[ WARN] [1690624558.486627876, 2000.393000000]: [Traj server]: ready.
[FSM]: state: WAIT_TARGET
wait for goal or trigger.
[FSM]: state: WAIT_TARGET
wait for goal or trigger.
[ WARN] [1690624562.649126978, 2004.345000000]: Waiting for trigger from [n3ctrl] from RC
double free or corruption (out)
[iris_0_ego_planner_node-3] process has died [pid 160986, exit code -6, cmd /home/wdc/catkin_ws/devel/lib/ego_planner/ego_planner_node ~odom_world:=/vins_estimator/odometry ~planning/bspline:=/xtdrone/iris_0/planning/bspline ~planning/data_display:=/xtdrone/iris_0/planning/data_display ~planning/broadcast_bspline_from_planner:=/broadcast_bspline ~planning/broadcast_bspline_to_planner:=/broadcast_bspline ~grid_map/odom:=/xtdrone/iris_0/vins_estimator/odometry ~grid_map/cloud:=/iris_0/pcl_render_node/points ~grid_map/pose:=/iris_0/camera_pose ~grid_map/depth:=/iris_0/realsense/depth_camera/depth/image_raw __name:=iris_0_ego_planner_node __log:=/home/wdc/.ros/log/ab9cb2a6-2df5-11ee-ac8b-f179362e13d2/iris_0_ego_planner_node-3.log].
log file: /home/wdc/.ros/log/ab9cb2a6-2df5-11ee-ac8b-f179362e13d2/iris_0_ego_planner_node-3*.log

bigsuperZZZX, 2023-07-30:
我看报错信息，跑的xtdrone的仿真环境吧？
如果用当前仓库下的原版代码编译和运行可以吗

wdczz, 2023-07-31:
Maybe,original code with a large probability is working.

wdczz, 2023-08-01:
> 我看报错信息，跑的xtdrone的仿真环境吧？ 如果用当前仓库下的原版代码编译和运行可以吗

Maybe,original code with a large probability is working.