ZJU-FAST-Lab/ego-planner #63: Integrating ego planner with Gazebo and px4

TanmayJain03, 2023-01-25:
What Ego planner has achieved is really appreciable. I am a ndergraduate college sophomore trying to replicate this paper integrated with gazebo and px4 but I am facing some issues.
 I am able to spawn, teleop control the drone, I am even able to visualize depth, pcl, odom in rviz but the drone isn't able to avoid obstacles even in simple environments, like it just rams straight into a wall then gets inverted.
It would be quite helpful if you could give some lead to progress or any resources that might be helpful.
Any help is deeply appreciated.
Regards.

bigsuperZZZX, 2023-01-29:
If you are Chinese, you can follow our step-by-step free course via this link: https://www.bilibili.com/video/BV1WZ4y167me
Otherwise please provide some videos or at least pictures showing the failures.

mdasimalam, 2024-07-31:
[image]
it doesn't showing obstacles

bigsuperZZZX, 2024-09-06:
It seems that you are not using this repository, which doesn't have sitl_px4.rviz as you show.
So raise issues in the repository which you are using.