ZJU-FAST-Lab/ego-planner #64: ZED instead of D435i

EhrazImam, 2023-01-30:
Hi @bigsuperZZZX  Hope you are doing well I want to use ZED camera instead of d435, As i am able to run it with D435 i want know what parameters i need to change apart from depth topic and intrinsic values.
Do i need to change something over here

Hi @ZhepeiWang Hope you are doing well I want to use ZED camera instead of d435, As i am able to run it with D435 i want know what parameters i need to change apart from depth topic and intrinsic values.
Do i need to change something over here

[Fast-Drone-250/src/planner/plan_manage/launch/advanced_param_exp.xml](https://github.com/ZJU-FAST-Lab/Fast-Drone-250/blob/8ff64274b161bb829653faf5946656658e7e2c9f/src/planner/plan_manage/launch/advanced_param_exp.xml#L103-L110)

Lines 103 to 110 in [8ff6427](https://github.com/ZJU-FAST-Lab/Fast-Drone-250/commit/8ff64274b161bb829653faf5946656658e7e2c9f)
  
 <param name="grid_map/use_depth_filter" value="true"/> 
 <param name="grid_map/depth_filter_tolerance" value="0.15"/> 
 <param name="grid_map/depth_filter_maxdist"   value="5.0"/> 
 <param name="grid_map/depth_filter_mindist"   value="0.2"/> 
 <param name="grid_map/depth_filter_margin"    value="2"/> 
 <param name="grid_map/k_depth_scaling_factor" value="1000.0"/> 
 <param name="grid_map/skip_pixel" value="2"/>
Lines 103 to 110 in [8ff6427](https://github.com/ZJU-FAST-Lab/Fast-Drone-250/commit/8ff64274b161bb829653faf5946656658e7e2c9f)
  
 <param name="grid_map/use_depth_filter" value="true"/> 
 <param name="grid_map/depth_filter_tolerance" value="0.15"/> 
 <param name="grid_map/depth_filter_maxdist"   value="5.0"/> 
 <param name="grid_map/depth_filter_mindist"   value="0.2"/> 
 <param name="grid_map/depth_filter_margin"    value="2"/> 
 <param name="grid_map/k_depth_scaling_factor" value="1000.0"/> 
 <param name="grid_map/skip_pixel" value="2"/>

bigsuperZZZX, 2023-01-31:
Make sure the resolution is 640*480. No more modifications are required.

EhrazImam, 2023-01-31:
I am using 720*480
so do i need to change something or i have to make it 640*480

bigsuperZZZX, 2023-01-31:
Correcting [this line](https://github.com/ZJU-FAST-Lab/ego-planner/blob/fcdfc5419dcc0c93105803e6f60b1b08b4e5152d/src/planner/plan_env/src/grid_map.cpp#L89) should be OK

closed 2023-02-23 by EhrazImam

FPSychotic, 2023-02-23:
If I'm using 320x240 should I change that line too?

bigsuperZZZX, 2023-03-16:
I think so.

DIY-create, 2023-07-06:
> Hi @bigsuperZZZX Hope you are doing well I want to use ZED camera instead of d435, As i am able to run it with D435 i want know what parameters i need to change apart from depth topic and intrinsic [...]

May I ask if your experiment was successful?

I also used a ZED2i camera to replace the realsense d435i.

 I replaced the topic of the depth image, changed the intrinsics of the depth image, and in the "ego planner/src/planner/plan_ Env/src/grid_ The map. cpp "code has modified the resolution to suit the resolution, and I have ensured that the depth image is normal.

But after I opened rviz, it still didn't display any maps. Do you know what other steps I need to take?

 Thank you very much