ZJU-FAST-Lab/ego-planner #65: Trajectory tracking control

zrd1234, 2023-02-02:
Hello, I subscribe /planning/pos_cmd to send to my robot, but I find some deviation from the trajectory of the planning department. Is there any good trajectory following strategy to solve it.

bigsuperZZZX, 2023-02-08:
It depends on the real robot you use. We use this high-level controller for precise trajectory tracking.

bigsuperZZZX, 2023-02-08:
This controller [https://github.com/ZJU-FAST-Lab/Fast-Drone-250/tree/master/src/realflight_modules/px4ctrl](https://github.com/ZJU-FAST-Lab/Fast-Drone-250/tree/master/src/realflight_modules/px4ctrl)

zrd1234, 2023-02-08:
> It depends on the real robot you use. We use this high-level controller for precise trajectory tracking.

Hello, I did not use UAV. In the simulation, PID was used for its trajectory tracking, but I did not find it. Would you please tell me where it is?

bigsuperZZZX, 2023-02-08:
Which simulation you refer to？

closed 2023-03-20 by zrd1234