ZJU-FAST-Lab/ego-planner #66: Global Trajectory doesn't pass through all waypoints

Kaapeine, 2023-02-16:
Hi, really grateful for the amazing work you've done with this project. I was trying to generate a trajectory using global waypoints, by passing them to the `planGlobalTrajWaypoints` method. But I have noticed that it ignores some of the waypoints, and the trajectory followed is not the same as our pre-generated minimum snap trajectory. This is in the absence of any obstacles so I would expect the drone to go through all the waypoints but it does not. What could be causing this issue? Thank you for all your support.

Kaapeine, 2023-02-21:
Update: I figured out this issue, it was because the planning horizon was too large for our trajectory. Reducing this fixed the problem. However, I'm facing another global trajectory related issue. The minimum snap trajectory generation fails if a large number of waypoints is given. Any idea how this is happening? In this case, The X Coefficients of the global trajector in global data was empty when I printed it out.

bigsuperZZZX, 2023-03-16:
I think if you are going to make the drone fly through a lot of waypoints, setting them in the launch file together is not a proper way. 
It is recommended to write a new node that always publishes the only next waypoint when the drone has arrived at the current one. It will be more robust and you can make sure the drone goes past the given waypoint before giving the next one.

Kaapeine, 2023-03-21:
To add the waypoints, I created a new subscriber which would call the `planGlobalTrajWaypoints` method in its callback. This currently works for 40-50 waypoints but beyond that, it fails because minimum snap trajectory generation fails. But thank you, I will try sending them one by one.

closed 2023-03-27 by Kaapeine