ZJU-FAST-Lab/ego-planner #68: PX4 Cntrl changes

EhrazImam, 2023-02-23:
As you can see over here https://github.com/ZJU-FAST-Lab/Fast-Drone-250/issues/33#issuecomment-1331634422
I want to know if we are using VIO then why we need mavros/imu/data as far i know that mavros/imu/data gives us acc,gyro and orientation and where as /mavros/imu/data_raw we only get acc and gyro.
If you are flying over VIO, why do you need the magnetometer?
Thankyou

bigsuperZZZX, 2023-03-16:
The VIO pose is computed relative to the world frame where the x-axis points to the forward direction when you start VINS.
However, the PX4 FCU update orientation is based on its own world frame. 
These two world frames are not aligned, especially in the yaw direction.
So we have to subscribe the mavros/imu/data to acquire the yaw difference between the world frames of VINS and the FCU, so that we can convert the quaternion command computed in the VINS world frame to the FCU world frame.

FPSychotic, 2023-03-16:
Can you point that part in the code, where is the quaternion conversion, I guess it is from NED to ENU , would be nice find it in the code if possible

bigsuperZZZX, 2023-03-18:
https://github.com/ZJU-FAST-Lab/Fast-Drone-250/blob/8ff64274b161bb829653faf5946656658e7e2c9f/src/realflight_modules/px4ctrl/src/controller.cpp#L51

EhrazImam, 2023-07-25:
@bigsuperZZZX how can we solve this if i dont want to pass compass data and just want it to run over /mavros/imu/data_raw and what will be the issue.

bigsuperZZZX, 2023-07-25:
You can [switch the state estimatior of PX4 from EKF to Q Estimator](https://docs.px4.io/main/en/advanced/switching_state_estimators.html#available-estimators) as this estimator can ignore the magnetometer readings.
To completely turn off the compass, set SYS_HAS_MAG to 0.

EhrazImam, 2023-07-26:
Thankyou @bigsuperZZZX for you reply.

This means i can take flight without having compass but as we are subscribing mavros/imu/data which is fusion of magnetometer and imu how i an going to subscribe if i am going to turn it of

bigsuperZZZX, 2023-07-28:
Magnetometer is not the necessarity for mavros/imu/data.

EhrazImam, 2023-07-31:
Thankyou @bigsuperZZZX

EhrazImam, 2023-07-31:
> The VIO pose is computed relative to the world frame where the x-axis points to the forward direction when you start VINS. However, the PX4 FCU update orientation is based on its own world frame. [...]

closed 2023-07-31 by EhrazImam

bigsuperZZZX, 2024-04-22:
@bzw-uestc you need to let pixhawk 6c transmit the orientation mavlink message of the same frequency to the IMU message, then you can get /mavros/imu/data back. Complementary filter estimator is not the problem.