ZJU-FAST-Lab/ego-planner #73: ego_planner error

Bruce-Si, 2023-03-20:
I got the following error when using ego_planner almost every time. Very occasionally it works well. 
[image]
The environment is unbuntu18.04 + melodic, I reinstalled the system and environment many times, it didn't work. Can anyone help me? 
By the way, the log file doesn't exit...

Swadine, 2023-03-20:
What is your map size?

Bruce-Si, 2023-03-20:
> What is your map size?

map_size_x = 100, map_size_y = 100, map_size_z = 5

bigsuperZZZX, 2023-03-24:
print out the odometry position to see if it goes divergence.

Bruce-Si, 2023-03-27:
@bigsuperZZZX The following is the odometry position, it seems ok since the map_size_x=100, map_size_y=100, map_size_z=5
[image]

bigsuperZZZX, 2023-04-05:
I have no idea about that. Never encounter such an error.

Swadine, 2023-04-15:
@Bruce-Si 
I have encountered this issue, somehow reducing the map size fixed it, but its not a feasible solution but then you'll have to shift to a better machine.

Bruce-Si, 2023-04-17:
Thank you. I tried to reduce the map size but it still did't work.😥Maybe I should try a different machine...

bigsuperZZZX, 2023-05-14:
You could try our recent stable version [https://github.com/ZJU-FAST-Lab/EGO-Planner-v2](https://github.com/ZJU-FAST-Lab/EGO-Planner-v2).

Bruce-Si, 2023-05-15:
> You could try our recent stable version https://github.com/ZJU-FAST-Lab/EGO-Planner-v2.

Thank you for your great work, I'll have a try.

closed 2023-06-29 by bigsuperZZZX