ZJU-FAST-Lab/ego-planner #86: Issue with 6c Mini

EhrazImam, 2023-08-30:
Dear @bigsuperZZZX i hope you are doing well,

Recently I purchased a 6c mini where i am trying to run px4ctrl as soon as i am giving takeoff command in initial testing it took off to 1 m height and after that it start dropping and in today's testing it started gaining height more than 1 m the firmware i am using is 1.13.3 even i rechecked vio but it was working fine drifting. 

https://github.com/ZJU-FAST-Lab/ego-planner/assets/88690390/dcce87fa-076a-46ea-aa43-1909819e8fcd

bigsuperZZZX, 2023-09-04:
Make sure the IMU vibration is adequately suppressed during flight. In particular, the acceleration noise along the z-axis should remain within the range of ±2m/s^2

EhrazImam, 2023-09-04:
@bigsuperZZZX  Thankyou for your reply is this the reason for continuously gaining the height even the height for takeoff is given for 1 m. If this so i will try to restucture my setup for reducing the noise.
Thankyou

bigsuperZZZX, 2023-09-04:
This is the most likely reason, but not all

EhrazImam, 2023-09-04:
Thankyou will check and let you know, I was confused weather it was structural problem or not.

bzw-uestc, 2023-12-08:
@EhrazImam hi, I meet the same question when using pixhawk 6c. I can't get /mavros/imu/data. Have you solved it?