ZJU-FAST-Lab/ego-planner #87: when I try to roslaunch multi_uva.launch, there is a repeat problem no odom, help.

ajmddzp, 2023-09-18:
[image]

bigsuperZZZX, 2023-09-20:
It means you didn't correctly connect your odometry topic.

bcwww, 2024-05-29:
Hello, have you solved the issue? I ran into the same problem.

bigsuperZZZX, 2024-06-27:
1. Use “rosnode list” command to list and identify the ROS node names of the planner and the drone simulator. 
2. Use "rosnode info xxx(the planner node name)" to see the odometry topic name the planner subscribes.
3. Use "rosnode info xxx(the simulator node name)" to see the odometry name published.
4. Remap the topic names to make them connected.