ZJU-FAST-Lab/ego-planner #89: roslaunch run_in_sim.launch后rviz中无人机没有出现

Mendeler, 2023-12-30:
我按照2. Standard Compilation进行了编译，然后在进行到Step 4. Run.的时候，没有出现预期的结果：
[image]
运行两个launch文件都没有报错，而且环境中的障碍物能够正常显示，但是没有无人机没有出现；
同时Fixed Frame 中没有TF数据，depth中的图像Image也没有数据，请问一下是有遗漏的步骤吗？还是rviz中需要进行哪些配置？

Mendeler, 2023-12-30:
而且我想通过对比ego-planner-swarm寻找原因，但是在ego-planner-swarm中也遇到了类似的问题：
[image]
没有报错，rviz中也没有显示出环境和无人机。
有人遇到同样的问题吗？

closed 2023-12-30 by Mendeler

Guangday, 2024-05-17:
我遇到了一样的问题，请问你是怎么解决的，所有的终端都正常的运行，UE4也是正常仿真，节点也正常运行，但是RVir中并没有任何内容。

bigsuperZZZX, 2024-06-27:
要是实在显示不了无人机，就把飞机的odometry消息显示出来吧，展示效果上差不多的