ZJU-FAST-Lab/ego-planner #92: error: no match for ‘operator=’

jierui778, 2024-02-05:
按照markdowm步骤构建发生如下错误
/home/jierui778/ego-planner/src/planner/traj_utils/include/traj_utils/polynomial_traj.h: In member function ‘double PolynomialTraj::getJerk()’:
/home/jierui778/ego-planner/src/planner/traj_utils/include/traj_utils/polynomial_traj.h:238:94: error: no match for ‘operator=’ (operand types are ‘Eigen::internal::enable_if<true, Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double> >::type {aka Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double>}’ and ‘double’)
               i * (i - 1) * (i - 2) * j * (j - 1) * (j - 2) * pow(ts, i + j - 5) / (i + j - 5);

bigsuperZZZX, 2024-04-23:
Eigen没装好吧，这里是最基础的Eigen矩阵赋值操作，不可能不支持的

Cyaeghas, 2024-10-22:
厚米听我说，这里的报错的代码，循环体里i和j用的是double，估计是下标类型不支持，所以要改成int。

bigsuperZZZX, 2024-11-20:
u1s1，雀食

closed 2024-11-20 by bigsuperZZZX