ZJU-FAST-Lab/ego-planner #93: error: no match for ‘operator=’

jierui778, 2024-02-05:
按照markdowm步骤构建发生如下错误
/home/jierui778/ego-planner/src/planner/traj_utils/include/traj_utils/polynomial_traj.h: In member function ‘double PolynomialTraj::getJerk()’:
/home/jierui778/ego-planner/src/planner/traj_utils/include/traj_utils/polynomial_traj.h:238:94: error: no match for ‘operator=’ (operand types are ‘Eigen::internal::enable_if<true, Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double> >::type {aka Eigen::IndexedView<Eigen::Matrix<double, -1, -1>, double, double>}’ and ‘double’)
               i * (i - 1) * (i - 2) * j * (j - 1) * (j - 2) * pow(ts, i + j - 5) / (i + j - 5);

closed 2024-02-15 by jierui778

lmhqsy, 2024-08-24:
请问解决了吗？我也遇到了同样的问题。

bigsuperZZZX, 2024-09-06:
Eigen版本不对吧，重装一下新版看看

lmhqsy, 2024-09-06:
重新装了也不行，我把代码里for循环中的double换成了int解决的编译