ZJU-FAST-Lab/ego-planner #94: catkin_make -DCMAKE_BUILD_TYPE=Release执行后报错

LJJcodeing, 2024-03-19:
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target nav_msgs_generate_messages_py
[  0%] Built target nav_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target nav_msgs_generate_messages_cpp
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_PositionCommand
[  0%] Built target nav_msgs_generate_messages_eus
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_Odometry
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_LQRTrajectory
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_AuxCommand
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_TRPYCommand
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_Gains
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_StatusData
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_PolynomialTrajectory
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_OutputData
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_Corrections
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_SO3Command
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_PPROutputData
[  0%] Built target nav_msgs_generate_messages_lisp
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libpcl_common.so', needed by '/home/ubuntu/Desktop/ego-planner/devel/lib/map_generator/random_forest'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:2651: uav_simulator/map_generator/CMakeFiles/random_forest.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  1%] Built target pose_utils
[  1%] Built target _quadrotor_msgs_generate_messages_check_deps_Serial
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libpcl_kdtree.so', needed by '/home/ubuntu/Desktop/ego-planner/devel/lib/mockamap/mockamap_node'.  Stop.
[  2%] Built target SO3Control
make[1]: *** [CMakeFiles/Makefile2:4275: uav_simulator/mockamap/CMakeFiles/mockamap_node.dir/all] Error 2
[  4%] Built target plan_env
[  4%] Built target std_msgs_generate_messages_lisp
[  4%] Built target std_msgs_generate_messages_nodejs
[  4%] Built target _multi_map_server_generate_messages_check_deps_MultiOccupancyGrid
[  4%] Built target _multi_map_server_generate_messages_check_deps_VerticalOccupancyGridList
[  4%] Built target _ego_planner_generate_messages_check_deps_DataDisp
[  4%] Built target _multi_map_server_generate_messages_check_deps_SparseMap3D
[  4%] Built target _multi_map_server_generate_messages_check_deps_MultiSparseMap3D
[  4%] Built target _ego_planner_generate_messages_check_deps_Bspline
make: *** [Makefile:141: all] Error 2
Invoking "make -j16 -l16" failed

所以大大能不能看看是哪里出问题了？

HuaYuXiao, 2024-04-29:
您遇到的这个问题是在尝试编译一个基于ROS（Robot Operating System）的项目时发生的。错误信息表明在编译过程中，make工具无法找到某些必要的库文件。具体来说，错误信息指出无法找到`libpcl_common.so`和`libpcl_kdtree.so`这两个库文件。

以下是解决这个问题的一些步骤：

1. **确认库是否存在**：首先，您需要确认这些库是否已经安装在您的系统上。可以使用`locate`命令或者`find`命令来搜索这些库文件是否存在于系统中。

2. **安装缺失的库**：如果这些库没有安装，您需要安装它们。这通常是通过您的Linux系统的包管理器来完成的。例如，对于Ubuntu系统，您可以使用`apt-get`来安装：

    ```bash
    sudo apt-get update
    sudo apt-get install libpcl-all
    ```

    或者，如果您使用的是其他发行版，可能需要使用不同的包管理器。

3. **链接库文件**：如果库文件已经安装，但是make工具找不到它们，可能是因为库文件没有链接到标准的库文件路径。您可以尝试创建一个符号链接到`/usr/lib/x86_64-linux-gnu/`目录：

    ```bash
    sudo ln -s /path/to/libpcl_common.so /usr/lib/x86_64-linux-gnu/libpcl_common.so
    sudo ln -s /path/to/libpcl_kdtree.so /usr/lib/x86_64-linux-gnu/libpcl_kdtree.so
    ```

    请将`/path/to/`替换为实际的库文件路径。

4. **重新运行make命令**：在解决了上述问题后，重新运行您的make命令：

    ```bash
    make -j16 -l16
    ```

5. **环境变量**：确保您的`LD_LIBRARY_PATH`环境变量包含了库文件的路径，这样系统才能找到它们。

6. **重新编译**：如果上述步骤都无法解决问题，您可能需要重新编译ROS相关的包，以确保所有的依赖都正确地被处理。

7. **检查版本兼容性**：确保您安装的PCL库版本与您的项目兼容。

如果您不熟悉这些步骤，或者不确定如何操作，可能需要寻求更专业的帮助，或者查看项目的文档和社区支持。

bigsuperZZZX, 2024-06-27:
听chatGPT的