ZJU-FAST-Lab/ego-planner #100: Can I control the plane myself when there are no obstacles?and trigger egoplanner when there is an obstacle

kaoxirui, 2024-06-04:
当我使用egoplanner时，如果没有障碍物时，我能不能使用自己的控制量控制飞机，然后在有障碍物时再把ego规划出的控制量发给飞机。

bigsuperZZZX, 2024-06-27:
可以，给traj_server节点中添加判断代码，用以切换控制指令的发送源

kaoxirui, 2024-06-27:
@bigsuperZZZX 但是我发现一个问题，不知道是不是求解速度的问题。比如我让飞机自己飞（不用ego的指令）并发送的定位给ego，比如0.5，1，1.5，2...这样，让飞机根据这个定位飞，egoplanner规划出来的位置信息跟不上我发送的位置，就好像没用到最新的odom位置信息

bigsuperZZZX, 2024-06-27:
你需要检查planner处在什么模式里，如果是REPLAN_TRAJ，则轨迹初值是从上一次的轨迹上取的，而非此刻的飞机odom。
代码：https://github.com/ZJU-FAST-Lab/ego-planner/blob/0835f963dcfadce156dbec8ab66b93930955a87c/src/planner/plan_manage/src/ego_replan_fsm.cpp#L273

kaoxirui, 2024-06-27:
@bigsuperZZZX 我看终端里一直都是在规划然后REPLAN_TRAJ，这个有办法设置吗，让他初始值使用此时的odom

bigsuperZZZX, 2024-06-27:
你在希望切换到轨迹规划的时候，可以给planner发一个goal，planner的下一次规划便会取odom作为初值

kaoxirui, 2024-06-27:
@bigsuperZZZX 因为我使用的控制方式通过ros publish目标点，是不是因为一直发同一个目标点，就会一直REPLAN_TRAJ

bigsuperZZZX, 2024-06-27:
接收到目标点后的第一次规划状态机进入的状态是GNE_NEW_TRAJ，此时从odom取规划初值，随后的规划才会进入REPLAN_TRAJ状态。

kaoxirui, 2024-06-27:
@bigsuperZZZX 按照您的意思，如果想切控制源，只能在想切换时发goal，如果一直发goal，就会进入REPLAN_TRAJ，在规划时不使用最新的odom

kaoxirui, 2024-06-27:
好的谢谢！

kaoxirui, 2024-06-27:
@bigsuperZZZX 好的谢谢！

kaoxirui, 2024-06-27:
@bigsuperZZZX 不好意思再打扰一下，我想问一下launch文件里预订给点的方式发送频率或者什么时候发送，我好像没找到发送频率
        
        
        <arg name="flight_type" value="2" />
        
        
        <arg name="point_num" value="1" />
        <arg name="point0_x" value="3" />
        <arg name="point0_y" value="0" />
        <arg name="point0_z" value="1" />
        <arg name="point1_x" value="-15.0" />
        <arg name="point1_y" value="0.0" />
        <arg name="point1_z" value="1.0" />
        <arg name="point2_x" value="15.0" />
        <arg name="point2_y" value="0.0" />
        <arg name="point2_z" value="1.0" />
        <arg name="point3_x" value="-15.0" />
        <arg name="point3_y" value="0.0" />
        <arg name="point3_z" value="1.0" />
        <arg name="point4_x" value="15.0" />
        <arg name="point4_y" value="0.0" />
        <arg name="point4_z" value="1.0" />

kaoxirui, 2024-06-27:
@bigsuperZZZX  不好意思再打扰一下，我想问一下launch文件里预订给点的方式发送频率或者什么时候发送，我好像没找到发送频率

> <arg name="flight_type" value="2" />
>
>
>        <arg name="point_num" value="1" />
>       <arg name="point0_x" value="3" />
>        <arg name="point0_y" value="0" />
>        <arg [...]

bigsuperZZZX, 2024-06-27:
不是通过launch来发送，而是你自己写新的节点来发送

kaoxirui, 2024-06-27:
@bigsuperZZZX 这个我清楚，我是想了解一下launch文件里的发送频率（和切换控制源那个问题没关系）

bigsuperZZZX, 2024-06-27:
launch文件是直接读取的，而不是发送接收的

kaoxirui, 2024-06-27:
@bigsuperZZZX 你好，我不太理解是在哪直接读取的，goal怎么发送给ego的。还有这个读取是到达目标点后读取下一个吗

closed 2024-06-29 by kaoxirui