ZJU-FAST-Lab/ego-planner #103: 3D LiDAR instead

bravewhh, 2024-07-17:
Great job! May I ask how I can integrate the ego planner code if my sensor is a 3D LiDAR instead of a binocular camera?

20221023lijia, 2024-07-22:
yes,you can.you need to change the  corresponding topic first.

bigsuperZZZX, 2024-07-28:
Publish point cloud in world frame to the topic [/grid_map/cloud](https://github.com/ZJU-FAST-Lab/ego-planner/blob/0835f963dcfadce156dbec8ab66b93930955a87c/src/planner/plan_env/src/grid_map.cpp#L120C50-L120C65)

Wangben1019, 2024-08-10:
> Publish point cloud in world frame to the topic [/grid_map/cloud](https://github.com/ZJU-FAST-Lab/ego-planner/blob/0835f963dcfadce156dbec8ab66b93930955a87c/src/planner/plan_env/src/grid_map.cpp#L120C50-L120C65)

请问一下具体更改方式是什么，加入我想要在真实无人机上使用Mid360+Fastlio输出里程计运行ego-planner，那么我是直接更改您指出的代码部分的点云和里程计的话题，还是在launch文件中做更改，对于部署真实无人机并做出上述更改，是使用本仓库的ego-planner更好些还是使用贵实验室开源的另一个仓库Fast-Drone-250中的代码更好一些，我发现这两个仓库有所不同

bigsuperZZZX, 2024-09-06:
1. 直接更改源码或更改launch均可
2. 用Fast-Drone-250

enordco, 2024-10-21:
Hi,

I am currently using px4ctrl (https://github.com/ZJU-FAST-Lab/Fast-Drone-250/tree/master/src/realflight_modules/px4ctrl) with the Fast LIO algorithm (https://github.com/hku-mars/FAST_LIO), running on Pixhawk Cube+ with PX4 firmware version 1.15.0 and a Livox Mid360 LiDAR for odometry.

When executing an automatic takeoff via the takeoff.sh script, the drone lifts off but starts oscillating along both the X and Z axes. Previously, I used a camera for odometry with px4ctrl, and it worked flawlessly. However, after switching to LiDAR, this oscillation issue has appeared.

I am attaching the flight log for reference. Kindly review and provide assistance. Thank you!

Flight Log:- https://drive.google.com/file/d/18RDbIVaIRgQszmVV3RXJ6CHgbowOQn8K/view?usp=drive_link

@bigsuperZZZX

bhaskar-glitch, 2024-10-21:
> 1. 直接更改源码或更改launch均可
> 
>     2. 用Fast-Drone-250

Hi,

I am currently using px4ctrl ( [https://github.com/ZJU-FAST-Lab/Fast-Drone-250/tree/master/src/realflight_modules/px4ctrl ](https://github.com/ZJU-FAST-Lab/Fast-Drone-250/tree/master/src/realflight_modules/px4ctrl)) with the Fast LIO algorithm ( [https://github.com/hku-mars/FAST_LIO ](https://github.com/hku-mars/FAST_LIO)), running on Pixhawk Cube+ with PX4 firmware version 1.15.0 and a Livox Mid360 LiDAR for odometry.

When executing an automatic takeoff via the takeoff.sh script, the drone lifts off but starts oscillating along both the X and Z axes. Previously, I used a camera for odometry with px4ctrl, and it worked flawlessly. However, after switching to LiDAR, this oscillation issue has appeared.

I am attaching the flight log for reference. Kindly review and provide assistance. Thank you!

Flight Log:- https://drive.google.com/file/d/18RDbIVaIRgQszmVV3RXJ6CHgbowOQn8K/view?usp=drive_link

@bigsuperZZZX @Wangben1019

bhaskar-glitch, 2024-11-20:
> Please check the frame rate of the new odometry topic from FAST LIO at first. If the output frame rate is very low (like about 10Hz), then you should write a Kalman Filter to fuse the high-rate IMU with the low-rate odometry from FAST LIO to get a high-rate odometry.

We are already getting 100hz of odometry from FAST LIO, still the px4 control is misbehaving

bigsuperZZZX, 2024-11-25:
Then there will be a lot of possible reasons leading to misbehaving. It's not easy to find out the real causes.

bhaskar-glitch, 2024-11-26:
> Then there will be a lot of possible reasons leading to misbehaving. It's not easy to find out the real causes.

Please let me know what else information should i provide to get the right help?

bigsuperZZZX, 2024-12-13:
You have to check the quailty of the positioning data(no vibration, no delay), the correctness of the framework definition between LIO and the controller (the px4ctrl package) by reading the px4ctrl code, the correctness of imu topic used by px4ctrl (which should be "imu/data" not "imu/data_raw"). Turn the feedback parameters in px4ctrl.