ZJU-FAST-Lab/ego-planner #107: How to set planning_horizon, why 7.5 meters?

lida2003, 2024-09-17:
Hi

I'm confused about this param " <arg name="planning_horizon" value="7.5" /> "

How to measure 7.5 meters? If it's something to do with "Field of View", why not 170 degree from camera spec?

bigsuperZZZX, 2024-09-18:
Maybe I didn't get the point.
What do you mean by "How to measure 7.5 meters?"

lida2003, 2024-09-18:
@bigsuperZZZX Thanks your time.

It's the configuration parameter, which is planning_horizon. I'm NOT sure what unit it is. It's NOT degree of FOV, pobably meters????

https://github.com/ZJU-FAST-Lab/ego-planner/blob/0835f963dcfadce156dbec8ab66b93930955a87c/src/planner/plan_manage/launch/simple_run.launch#L38-L39

bigsuperZZZX, 2024-09-19:
The unit is meter. "sensing horizon" is the distance that depth sensor can reach with acceptable accuracy.

lida2003, 2024-09-19:
Alright, at least we can confirm that the unit is meters. Now from the code’s perspective, it should be 1.5 times a certain specification parameter. Logically, the specification parameter is 5 meters. I would like to know which parameter in the specification sheet corresponds to this 5 meters?