ZJU-FAST-Lab/ego-planner #114: 请问，当我使用fastlio2配合egoplanner,启动rviz后，为什么odometry的坐标系一直在上下跳动

avicomac, 2025-03-04:
[image]

bigsuperZZZX, 2025-03-08:
可以检查下rviz可视化用的odometry是哪个节点发出来的，以便朔源问题。
此外，z轴的这种跳动比较像错误处理重力加速度导致的。

XXLiu-HNU, 2025-03-26:
> 可以检查下rviz可视化用的odometry是哪个节点发出来的，以便朔源问题。 此外，z轴的这种跳动比较像错误处理重力加速度导致的。

 @bigsuperZZZX 周博你好，我在使用Fast-LIO2 和px4ctrl的时候也发现了无人机会一直在Z轴上上下抖动。我设置的起飞高度是1.0m，飞机实际上会不断的在1m左右徘徊。从RVIZ上来看定位应该没飘。是因为输出的里程计没有速度的原因吗。这是我的测试环境

[image]

这是我的测试视频
 
https://github.com/user-attachments/assets/19dbe5a2-4119-44c3-a010-ff05f116e910

avicomac, 2025-03-31:
> > 可以检查下rviz可视化用的odometry是哪个节点发出来的，以便朔源问题。 此外，z轴的这种跳动比较像错误处理重力加速度导致的。
> 
> [@bigsuperZZZX](https://github.com/bigsuperZZZX) 周博你好，我在使用Fast-LIO2 和px4ctrl的时候也发现了无人机会一直在Z轴上上下抖动。我设置的起飞高度是1.0m，飞机实际上会不断的在1m左右徘徊。从RVIZ上来看定位应该没飘。是因为输出的里程计没有速度的原因吗。这是我的测试环境
> 
> [image]
> 
> 这是我的测试视频
> video-2025-03-26_23.11.53.mp4

当我在iris无人机上添加mid360雷达，以及使用mid360仿真插件实现定位时，也会出现这个现象，奇怪的是，当我放在另一台无人机上，没有出现这个问题

XXLiu-HNU, 2025-03-31:
> 当我在iris无人机上添加mid360雷达，以及使用mid360仿真插件实现定位时，也会出现这个现象，奇怪的是，当我放在另一台无人机上，没有出现这个问题

神奇，但是我在做实物实验的时候也会出现z 轴抖动的问题。仿真可能是没以后速度输入，我增加了速度输出就好很多了

avicomac, 2025-03-31:
请教老哥，除了px4_ctrl还有别的比较好的开源的控制器吗，我觉得px4_ctrl太复杂了

once
***@***.***

      

  
 3289632212.jpg (view on web)
  
  

    
  
 3289632212.jpg (view on web)

XXLiu-HNU, 2025-03-31:
> 请教老哥，除了px4_ctrl还有别的比较好的开源的控制器吗，我觉得px4_ctrl太复杂了

其实复杂倒也不复杂，核心的PID控制部分只有一个对速度和位置的P，反倒是比较简单。 主要复杂的部分是状态机的部分，但其实稍微阅读一下会发现其实挺好用。 如果觉得px4ctrl复杂可以直接找个PID控制器也行

XXLiu-HNU, 2025-04-07:
> 请教佬哥，这个cmu的控制器有了解过吗，[https://github.com/Zhefan-Xu/tracking_controller；我不知道怎么针对我自己的无人机调参，实现好的追踪效果。我自己整了一个简单的控制器，但是效果不好，我把gazebo真值给无人机作为mavros的视觉输入仍然是差不多的结果

这个控制器我也测试了，这里边的悬停油门的参数总感觉有点调不好，px4ctrl 的油门是在线估计的。我测试的效果比px4ctrl还差一点，可能是参数调整的问题。如果用PID控制器本质上就是调参，调到一个合适的参数应该就可以。

avicomac, 2025-04-07:
> 请教佬哥，这个cmu的控制器有了解过吗，[https://github.com/Zhefan-Xu/tracking_controller]；我不知道怎么针对我自己的无人机调参，实现好的追踪效果。我自己整了一个简单的控制器，但是效果不好，我把gazebo真值给无人机作为mavros的视觉输入仍然是差不多的结果
>
> 不过px4ctrl我实物实验的时候总是炸鸡，无人机自动起飞之后直接冲上天，最近我也在找好用的控制，如果大佬找到比较好用的麻烦分享一下~
>
>
> 请教佬哥，这个cmu的控制器有了解过吗，[https://github.com/Zhefan-Xu/tracking_controller]；我不知道怎么针对我自己的无人机调参，实现好的追踪效果。我自己整了一个简单的控制器，但是效果不好，我把gazebo真值给无人机作为mavros的视觉输入仍然是差不多的结果
>
> 不过px4ctrl我实物实验的时候总是炸鸡，无人机自动起飞之后直接冲上天，最近我也在找好用的控制，如果大佬找到比较好用的麻烦分享一下~

好的老哥，我也是冲天，差点把雷达摔坏

once
***@***.***

avicomac, 2025-04-07:
佬哥，cmu这个我调了一下，将就能用，速度不能快，悬停油门好像是不好用，跟踪第一下会掉高度然后再追

once
***@***.***

XXLiu-HNU, 2025-04-13:
> 佬哥，cmu这个我调了一下，将就能用，速度不能快，悬停油门好像是不好用，跟踪第一下会掉高度然后再追

如果是使用XTDrone仿真环境的话，第一下掉是因为XTDrone的通讯模块也会发送控制质量，导致两个指令冲突

XXLiu-HNU, 2025-06-09:
> 老哥，像你请教一下，如果我不用上层控制器，直接把规划器的输出的位置或者速度传到底层px4跟踪的话，能跟踪吗？会有什么后果
> […](#)

可以跟踪，只是如果速度比较快的话可能跟踪误差会有点大

avicomac, 2025-06-09:
老哥，那为啥我上层用一个pid，就基于位置误差解算出来一个速度喂到px4它的跟踪效果就会好一点？px4和这个上层的pid不都是pid吗，我有点没搞明白原理

once
***@***.***

XXLiu-HNU, 2025-06-10:
> 老哥，那为啥我上层用一个pid，就基于位置误差解算出来一个速度喂到px4它的跟踪效果就会好一点？px4和这个上层的pid不都是pid吗，我有点没搞明白原理

我的理解是，基于位置误差解算出来一个速度喂到px4相当于在px4自己的pid前边又串联了一个p，给px4输入的期望是一个“更合理”的，所以px4内部的pid处理起来效果也同样会好一些

bigsuperZZZX, 2025-06-21:
雀食
上层控制器的关键还是如何计算控制前馈，而且至少得是姿态前馈，如果角速度前馈就更好了