ZJU-FAST-Lab/ego-planner #118: vtk library error during catkin_make

HuaYuXiao, 2025-05-17:
```
==> Processing catkin package: 'ego_planner'
==> Creating build directory: 'build_isolated/ego_planner'
==> Building with env: '/home/easonhua/easondrone_ws/devel_isolated/traj_utils/env.sh'
==> cmake /home/easonhua/easondrone_ws/src/plan/ego-planner -DCATKIN_DEVEL_PREFIX=/home/easonhua/easondrone_ws/devel_isolated/ego_planner -DCMAKE_INSTALL_PREFIX=/home/easonhua/easondrone_ws/install_isolated -G Unix Makefiles in '/home/easonhua/easondrone_ws/build_isolated/ego_planner'
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Checking for module 'eigen3'
--   Found eigen3, version 3.3.7
-- Found Eigen: /usr/include/eigen3 (Required is at least version "3.1") 
-- Eigen found (include: /usr/include/eigen3, version: 3.3.7)
-- Found Boost: /usr/include (found suitable version "1.71.0", minimum required is "1.55.0") found components: system filesystem date_time iostreams regex 
-- Checking for module 'flann'
--   Found flann, version 1.9.1
-- Found FLANN: /usr/lib/x86_64-linux-gnu/libflann_cpp.so  
-- The imported target "vtkParseOGLExt" references the file
   "/usr/bin/vtkParseOGLExt-7.1"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "pvtk" references the file
   "/usr/bin/pvtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.23
-- Found USB_10: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- Found OpenNI: /usr/lib/libOpenNI.so  
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- Found OpenNI2: /usr/lib/libOpenNI2.so  
-- OpenNI2 found (include: /usr/include/openni2, lib: /usr/lib/libOpenNI2.so)
[... 142 lines cut ...]
/usr/bin/ld: /home/easonhua/easondrone_ws/devel_isolated/path_searching/lib/libpath_searching.so: undefined reference to `vtkObjectFactoryRegistryCleanup::vtkObjectFactoryRegistryCleanup()'
/usr/bin/ld: /home/easonhua/easondrone_ws/devel_isolated/path_searching/lib/libpath_searching.so: undefined reference to `vtkObjectFactoryRegistryCleanup::~vtkObjectFactoryRegistryCleanup()'
/usr/bin/ld: /home/easonhua/easondrone_ws/devel_isolated/path_searching/lib/libpath_searching.so: undefined reference to `vtkRenderingOpenGL2_AutoInit_Destruct()'
/usr/bin/ld: /home/easonhua/easondrone_ws/devel_isolated/path_searching/lib/libpath_searching.so: undefined reference to `vtkDebugLeaksManager::~vtkDebugLeaksManager()'
/usr/bin/ld: /home/easonhua/easondrone_ws/devel_isolated/path_searching/lib/libpath_searching.so: undefined reference to `vtkRenderingFreeType_AutoInit_Destruct()'
/usr/bin/ld: /home/easonhua/easondrone_ws/devel_isolated/path_searching/lib/libpath_searching.so: undefined reference to `vtkInteractionStyle_AutoInit_Destruct()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/traj_server.dir/build.make:156: /home/easonhua/easondrone_ws/devel_isolated/ego_planner/lib/ego_planner/traj_server] Error 1
make[1]: *** [CMakeFiles/Makefile2:1425: CMakeFiles/traj_server.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Linking CXX executable /home/easonhua/easondrone_ws/devel_isolated/ego_planner/lib/ego_planner/ego_planner_node
/usr/bin/ld: CMakeFiles/ego_planner_node.dir/src/ego_planner_node.cpp.o: undefined reference to symbol '_Z37vtkRenderingOpenGL2_AutoInit_Destructv'
/usr/bin/ld: /lib/x86_64-linux-gnu/libvtkRenderingOpenGL2-7.1.so.7.1p: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/ego_planner_node.dir/build.make:186: /home/easonhua/easondrone_ws/devel_isolated/ego_planner/lib/ego_planner/ego_planner_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:547: CMakeFiles/ego_planner_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
<== Failed to process package 'ego_planner': 
  Command '['/home/easonhua/easondrone_ws/devel_isolated/traj_utils/env.sh', 'make', '-j22', '-l22']' returned non-zero exit status 2.

Reproduce this error by running:
==> cd /home/easonhua/easondrone_ws/build_isolated/ego_planner && /home/easonhua/easondrone_ws/devel_isolated/traj_utils/env.sh make -j22 -l22

Command failed, exiting.
```

HuaYuXiao, 2025-05-18:
https://answers.ros.org/question/261754/

bigsuperZZZX, 2025-06-21:
🤔🤔🤔