ZJU-FAST-Lab/ego-planner #122: 规划轨迹重规划问题

DouDaoMAX, 2025-08-16:
各位好，我最近想拿egoplanner_ros2在仿真无人车下去进行路径规划，mid360+fast_lio2,fastlio2提供其点云信息和里程计，然后octomap提供grip_map,加一个去除地面点云，一切准备就绪，运行ego_planner手动标点，它在仿真中的轨迹会一直乱跳，然后在终端显示规划失败重新规划，一开始是推测omdo的物理高度过低，后面把它调高了，也是会出现这个问题，我想问问有什么解决方法
`[ego_planner_node-1] [drone -1 replan 20284]==============================================
[ego_planner_node-1] multi-trajs=1
[ego_planner_node-1] iter=5,time(ms)=0.011,rebound.
[ego_planner_node-1] traj 1 failed.
[ego_planner_node-1] plan_success=0
[ego_planner_node-1] refine_success=0
[ego_planner_node-1] 
[ego_planner_node-1] [drone -1 replan 20285]==============================================
[ego_planner_node-1] multi-trajs=1
[ego_planner_node-1] iter=5,time(ms)=0.01,rebound.
[ego_planner_node-1] traj 1 failed.
[ego_planner_node-1] plan_success=0
[ego_planner_node-1] refine_success=0
[ego_planner_node-1] 
[ego_planner_node-1] [drone -1 replan 20286]==============================================
[ego_planner_node-1] multi-trajs=1
[ego_planner_node-1] iter=5,time(ms)=0.0099,rebound.
[ego_planner_node-1] traj 1 failed.
[ego_planner_node-1] plan_success=0
[ego_planner_node-1] refine_success=0
[ego_planner_node-1] 
[ego_planner_node-1] [drone -1 replan 20287]==============================================
[ego_planner_node-1] multi-trajs=1
[ego_planner_node-1] iter=5,time(ms)=0.009,rebound.
[ego_planner_node-1] traj 1 failed.
[ego_planner_node-1] plan_success=0
[ego_planner_node-1] refine_success=0
[ego_planner_node-1] 
[ego_planner_node-1] [drone -1 replan 20288]==============================================
[ego_planner_node-1] multi-trajs=1
[ego_planner_node-1] iter=5,time(ms)=0.011,rebound.
[ego_planner_node-1] traj 1 failed.
[ego_planner_node-1] plan_success=0
[ego_planner_node-1] refine_success=0
[ego_planner_node-1] [ERROR] [1755352917.667681567] [check_collision_and_rebound]: ERROR! the drone is in obstacle. This should not happen.
[ego_planner_node-1] [WARN] [1755352917.667683412] [check_collision_and_rebound]: WARN! terminal point of the current trajectory is in obstacle, skip this planning.
[ego_planner_node-1] [ERROR] [1755352917.667740672] [check_collision_and_rebound]: ERROR! the drone is in obstacle. This should not happen.
[ego_planner_node-1] [WARN] [1755352917.667742281] [check_collision_and_rebound]: WARN! terminal point of the current trajectory is in obstacle, skip this planning.
[ego_planner_node-1] [ERROR] [1755352917.667789986] [check_collision_and_rebound]: ERROR! the drone is in obstacle. This should not happen.
[ego_planner_node-1] [WARN] [1755352917.667791676] [check_collision_and_rebound]: WARN! terminal point of the current trajectory is in obstacle, skip this planning.
[ego_planner_node-1] [ERROR] [1755352917.667837226] [check_collision_and_rebound]: ERROR! the drone is in obstacle. This should not happen.
[ego_planner_node-1] [WARN] [1755352917.667838805] [check_collision_and_rebound]: WARN! terminal point of the current trajectory is in obstacle, skip this planning.
[ego_planner_node-1] [ERROR] [1755352917.667893343] [check_collision_and_rebound]: ERROR! the drone is in obstacle. This should not happen.
[ego_planner_node-1] [WARN] [1755352917.667894959] [check_collision_and_rebound]: WARN! terminal point of the current trajectory is in obstacle, skip this planning.
[ego_planner_node-1] [ERROR] [1755352917.667936767] [Coord2Index]: Ran out of pool, index=100 19 23, POOL_SIZE=100 100 100
[ego_planner_node-1] [ERROR] [1755352917.667938996] [AstarSearch]: Unable to handle the initial or end point, force return!
[ego_planner_node-1] [ERROR] [1755352917.667940882] [initControlPoints]: a star error, force return!
[ego_planner_node-1] [ERROR] [1755352917.667948896] [check_collision_and_rebound]: ERROR! the drone is in obstacle. This should not happen.
[ego_planner_node-1] [ERROR] [1755352917.667951982] [Coord2Index]: Ran out of pool, index=100 20 24, POOL_SIZE=100 100 100
[ego_planner_node-1] [ERROR] [1755352917.667954079] [AstarSearch]: Unable to handle the initial or end point, force return!
[ego_planner_node-1] [ERROR] [1755352917.667955789] [check_collision_and_rebound]: a star error
[ego_planner_node-1] [ERROR] [1755352917.668002491] [check_collision_and_rebound]: ERROR! the drone is in obstacle. This should not happen.
[ego_planner_node-1] [WARN] [1755352917.668004336] [check_collision_and_rebound]: WARN! terminal point of the current trajectory is in obstacle, skip this planning.
[ego_planner_node-1] 
[ego_planner_node-1] [drone -1 replan 20289]==============================================
[ego_planner_node-1] multi-trajs=1
[ego_planner_node-1] iter=5,time(ms)=0.052,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.011,rebound.
[ego_planner_node-1] traj 1 failed.
[ego_planner_node-1] plan_success=0
[ego_planner_node-1] refine_success=0
[ego_planner_node-1] [FSM]: from GEN_NEW_TRAJ to GEN_NEW_TRAJ
[ego_planner_node-1] [ERROR] [1755352917.678161676] [Coord2Index]: Ran out of pool, index=100 20 37, POOL_SIZE=100 100 100
[ego_planner_node-1] [ERROR] [1755352917.678173439] [AstarSearch]: Unable to handle the initial or end point, force return!
[ego_planner_node-1] [ERROR] [1755352917.678176910] [initControlPoints]: a star error, force return!
[ego_planner_node-1] [ERROR] [1755352917.678192325] [check_collision_and_rebound]: ERROR! the drone is in obstacle. This should not happen.
[ego_planner_node-1] [WARN] [1755352917.678194637] [check_collision_and_rebound]: WARN! terminal point of the current trajectory is in obstacle, skip this planning.
[ego_planner_node-1] 
[ego_planner_node-1] [drone -1 replan 20290]==============================================
[... 37 lines cut ...]
[ego_planner_node-1] [ERROR] [1755352917.678501436] [AstarSearch]: Unable to handle the initial or end point, force return!
[ego_planner_node-1] [ERROR] [1755352917.678516564] [initControlPoints]: a star error, force return!
[ego_planner_node-1] [ERROR] [1755352917.678526151] [check_collision_and_rebound]: ERROR! the drone is in obstacle. This should not happen.
[ego_planner_node-1] [ERROR] [1755352917.678528838] [Coord2Index]: Ran out of pool, index=100 68 34, POOL_SIZE=100 100 100
[ego_planner_node-1] [ERROR] [1755352917.678531114] [AstarSearch]: Unable to handle the initial or end point, force return!
[ego_planner_node-1] [ERROR] [1755352917.678532928] [check_collision_and_rebound]: a star error
[ego_planner_node-1] [ERROR] [1755352917.678568046] [check_collision_and_rebound]: ERROR! the drone is in obstacle. This should not happen.
[ego_planner_node-1] [ERROR] [1755352917.678570452] [Coord2Index]: Ran out of pool, index=100 68 34, POOL_SIZE=100 100 100
[ego_planner_node-1] [ERROR] [1755352917.678572370] [AstarSearch]: Unable to handle the initial or end point, force return!
[ego_planner_node-1] [ERROR] [1755352917.678573826] [check_collision_and_rebound]: a star error
[ego_planner_node-1] [ERROR] [1755352917.678584319] [check_collision_and_rebound]: ERROR! the drone is in obstacle. This should not happen.
[ego_planner_node-1] [ERROR] [1755352917.678586580] [Coord2Index]: Ran out of pool, index=100 68 34, POOL_SIZE=100 100 100
[ego_planner_node-1] [ERROR] [1755352917.678588385] [AstarSearch]: Unable to handle the initial or end point, force return!
[ego_planner_node-1] [ERROR] [1755352917.678589816] [check_collision_and_rebound]: a star error
[ego_planner_node-1] [ERROR] [1755352917.678599541] [check_collision_and_rebound]: ERROR! the drone is in obstacle. This should not happen.
[ego_planner_node-1] [ERROR] [1755352917.678601804] [Coord2Index]: Ran out of pool, index=100 68 34, POOL_SIZE=100 100 100
[ego_planner_node-1] [ERROR] [1755352917.678603576] [AstarSearch]: Unable to handle the initial or end point, force return!
[ego_planner_node-1] [ERROR] [1755352917.678605210] [check_collision_and_rebound]: a star error
`

https://github.com/user-attachments/assets/b75d1d2a-9ef3-43bf-9275-9f13adcf34a1

[image]
[image]
[image]

DouDaoMAX, 2025-08-16:
[ego_planner_node-1] [drone -1 replan 761433]==============================================
[ego_planner_node-1] multi-trajs=1
[ego_planner_node-1] iter=5,time(ms)=0.009338,rebound.
[ego_planner_node-1] traj 1 failed.
[ego_planner_node-1] plan_success=0
[ego_planner_node-1] refine_success=0
[ego_planner_node-1] 
[ego_planner_node-1] [drone -1 replan 761434]==============================================
[ego_planner_node-1] multi-trajs=1
[ego_planner_node-1] iter=5,time(ms)=0.009303,rebound.
[ego_planner_node-1] traj 1 failed.
[ego_planner_node-1] plan_success=0
[ego_planner_node-1] refine_success=0
[ego_planner_node-1] 
[ego_planner_node-1] [drone -1 replan 761435]==============================================
[ego_planner_node-1] multi-trajs=1
[ego_planner_node-1] iter=5,time(ms)=0.009163,rebound.
[ego_planner_node-1] traj 1 failed.
[ego_planner_node-1] plan_success=0
[ego_planner_node-1] refine_success=0
[ego_planner_node-1] 
[ego_planner_node-1] [drone -1 replan 761436]==============================================
[ego_planner_node-1] multi-trajs=1
[ego_planner_node-1] iter=5,time(ms)=0.012977,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.014908,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012653,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012339,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012309,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012592,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012507,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012183,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012185,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012199,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012212,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012375,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.013767,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012274,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012288,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012649,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012349,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012314,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.01237,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012257,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012508,rebound.
[ego_planner_node-1] traj 1 failed.
[ego_planner_node-1] plan_success=0
[ego_planner_node-1] refine_success=0
[ego_planner_node-1] [ERROR] [1755355296.794860591] [check_collision_and_rebound]: ERROR! the drone is in obstacle. This should not happen.
[ego_planner_node-1] [ERROR] [1755355296.794863041] [Coord2Index]: Ran out of pool, index=-1 12 36, POOL_SIZE=100 100 100
[ego_planner_node-1] [ERROR] [1755355296.794864814] [AstarSearch]: Unable to handle the initial or end point, force return!
[ego_planner_node-1] [ERROR] [1755355296.794866191] [check_collision_and_rebound]: a star error
[ego_planner_node-1] [ERROR] [1755355296.794902372] [Coord2Index]: Ran out of pool, index=-1 31 36, POOL_SIZE=100 100 100
[ego_planner_node-1] [ERROR] [1755355296.794904421] [AstarSearch]: Unable to handle the initial or end point, force return!
[ego_planner_node-1] [ERROR] [1755355296.794905908] [initControlPoints]: a star error, force return!
[ego_planner_node-1] [ERROR] [1755355296.794912774] [check_collision_and_rebound]: ERROR! the drone is in obstacle. This should not happen.
[ego_planner_node-1] [WARN] [1755355296.794914581] [check_collision_and_rebound]: WARN! terminal point of the current trajectory is in obstacle, skip this planning.
[ego_planner_node-1] [ERROR] [1755355296.794950868] [Coord2Index]: Ran out of pool, index=-1 1 13, POOL_SIZE=100 100 100
[ego_planner_node-1] [ERROR] [1755355296.794955306] [AstarSearch]: Unable to handle the initial or end point, force return!
[ego_planner_node-1] [ERROR] [1755355296.794956793] [initControlPoints]: a star error, force return!
[ego_planner_node-1] 
[ego_planner_node-1] [drone -1 replan 761437]==============================================
[ego_planner_node-1] multi-trajs=1
[ego_planner_node-1] iter=5,time(ms)=0.009446,rebound.
[ego_planner_node-1] traj 1 failed.
[ego_planner_node-1] plan_success=0
[ego_planner_node-1] refine_success=0
[ego_planner_node-1] 
[ego_planner_node-1] [drone -1 replan 761438]==============================================
[ego_planner_node-1] multi-trajs=1
[ego_planner_node-1] [ERROR] [1755355296.794963641] [check_collision_and_rebound]: ERROR! the drone is in obstacle. This should not happen.
[... 219 lines cut ...]
[ego_planner_node-1] iter=5,time(ms)=0.011974,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.013508,rebound.
[ego_planner_node-1] [ERROR] [1755355296.805898279] [Coord2Index]: Ran out of pool, index=-1 36 41, POOL_SIZE=100 100 100
[ego_planner_node-1] [ERROR] [1755355296.805901721] [AstarSearch]: Unable to handle the initial or end point, force return!
[ego_planner_node-1] [ERROR] [1755355296.805903112] [check_collision_and_rebound]: a star error
[ego_planner_node-1] [ERROR] [1755355296.805910590] [check_collision_and_rebound]: ERROR! the drone is in obstacle. This should not happen.
[ego_planner_node-1] [ERROR] [1755355296.805912833] [Coord2Index]: Ran out of pool, index=-1 36 41, POOL_SIZE=100 100 100
[ego_planner_node-1] [ERROR] [1755355296.805914594] [AstarSearch]: Unable to handle the initial or end point, force return!
[ego_planner_node-1] [ERROR] [1755355296.805915975] [check_collision_and_rebound]: a star error
[ego_planner_node-1] iter=5,time(ms)=0.01365,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012155,rebound.
[... cut ...]
[ego_planner_node-1] [ERROR] [1755355296.805938602] [Coord2Index]: Ran out of pool, index=-1 36 41, POOL_SIZE=100 100 100
[ego_planner_node-1] [ERROR] [1755355296.805940366] [AstarSearch]: Unable to handle the initial or end point, force return!
[ego_planner_node-1] [ERROR] [1755355296.805941764] [check_collision_and_rebound]: a star error
[ego_planner_node-1] iter=5,time(ms)=0.012184,rebound.
[ego_planner_node-1] iter=5,time(ms)=0.012184,rebound.
[ego_planner_node-1] traj 1 failed.
[ego_planner_node-1] plan_success=0
[ego_planner_node-1] refine_success=0
[ego_planner_node-1]

giangdao1402, 2025-10-05:
Hi @DouDaoMAX, did you figure out the solution ?